tinygo-org/tinygo #301: panic: todo: return value from coroutine

deadprogram, 2019-04-24:
I discovered a case when I receive the error `panic: todo: return value from coroutine`.

```shell
$ tinygo build -o flash.hex -target pca10040 examples/blinky2
panic: todo: return value from coroutine

goroutine 1 [running]:
github.com/tinygo-org/tinygo/compiler.(*Compiler).markAsyncFunctions(0xc0000e0b00, 0xc00121ad00, 0x121ad000041ef68, 0xd9c1c0)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/compiler/goroutine-lowering.go:397 +0x495c
github.com/tinygo-org/tinygo/compiler.(*Compiler).LowerGoroutines(0xc0000e0b00, 0x11fd940, 0x1206d48)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/compiler/goroutine-lowering.go:109 +0x40
github.com/tinygo-org/tinygo/compiler.(*Compiler).Optimize(0xc0000e0b00, 0x2, 0x2, 0x5, 0x0, 0x0)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/compiler/optimizer.go:73 +0x4fb
main.Compile(0x7fff5a8178cb, 0x10, 0x7fff5a8178b0, 0x9, 0xc0000e02c0, 0xc001ad3f10, 0xc001ad3da0, 0x0, 0x0)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/main.go:132 +0x2e67
main.Build(0x7fff5a8178cb, 0x10, 0x7fff5a8178b0, 0x9, 0x7fff5a8178c2, 0x8, 0xc0000b1f10, 0xc00008c7e0, 0x3d)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/main.go:284 +0xe2
main.main()
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/main.go:565 +0xbbe
```

Here is the code that can reproduce. It is a modified version of blinky2:

```go
package main

// This blinky is a bit more advanced than blink1, with two goroutines running
// at the same time and blinking a different LED. The delay of led2 is slightly
// less than half of led1, which would be hard to do without some sort of
// concurrency.

import (
	"machine"
	"time"
)

func main() {
	go led1()
	led2()
}

func led1() {
	led := machine.GPIO{machine.LED}
	led.Configure(machine.GPIOConfig{Mode: machine.GPIO_OUTPUT})
	for {
		println("+")
		led.Low()
		time.Sleep(time.Millisecond * 1000)

		println("-")
		led.High()
		time.Sleep(time.Millisecond * 1000)
	}
}

func led2() {
	led := getLED()

	for {
		println("  +")
		led.Low()
		time.Sleep(time.Millisecond * 420)

		println("  -")
		led.High()
		time.Sleep(time.Millisecond * 420)
	}
}

func getLED() *machine.GPIO {
	l := machine.GPIO{machine.LED2}
	l.Configure(machine.GPIOConfig{Mode: machine.GPIO_OUTPUT})
	time.Sleep(time.Millisecond * 1000)
	return &l
}
```

aykevl, 2019-04-24:
Yes, this is not yet implemented. Returning a pointer shouldn't be too difficult to add.

deadprogram, 2019-05-05:
This has been addressed in the `dev` branch. Now closing.

PR #317 (2019-05-05): compiler: support returning values from async functions
This is implemented as follows:

  * The parent coroutine allocates space for the return value in its
    frame and stores a pointer to this frame in the parent coroutine
    handle.
  * The child coroutine obtains the alloca from its parent using the
    parent coroutine handle. It then stores the result value there.
  * The parent value reads the data from the alloca on resumption.

The current implementation is not very efficient but it's a start.

Fixes #301.
closed 2019-05-05 by deadprogram