tinygo-org/tinygo #305: TinyGo can't find wasm-ld

johanbrandhorst, 2019-04-25:
Running latest TinyGo from `dev`, with all dependencies installed, I get the following error compiling some wasm:

```
tinygo build -o ./html/test.wasm -target wasm --no-debug ./js/main.go
error: failed to link /tmp/tinygo332217407/main: exec: "wasm-ld-8": executable file not found in $PATH
make: *** [Makefile:10: tinygo] Error 1
```

After symlinking `/usr/bin/wasm-ld` to `/usr/bin/wasm-ld-8` it works. I think we should support both paths.

johanbrandhorst, 2019-04-25:
This seems to be intentionally targetting `wasm-ld-8` only on linux. Is this the name used on Ubuntu and Debian?

https://github.com/tinygo-org/tinygo/blob/fa5df4f524c3b4f15360c2da964932692e3ab4af/commands.go#L9

johanbrandhorst, 2019-04-25:
We should presumably switch this over at some point? Can we remove this translation?

aykevl, 2019-04-25:
Well, Debian only has `wasm-ld-8` and not `wasm-ld`. I would propose to add a fallback mechanism, for example by listing multiple alternatives:

```go
"wasm-ld": []string{"wasm-ld-8", "wasm-ld"},
```

I have a partial patch for that lying around here somewhere...

aykevl, 2019-05-03:
@johanbrandhorst can you try #323?

johanbrandhorst, 2019-05-03:
Yep, works for me!

aykevl, 2019-05-05:
This has been fixed in #323.

closed 2019-05-05 by aykevl

talentlessguy, 2020-05-25:
For some reason it keeps happening :/

```
➜ tinygo build -o build/out.wasm -target wasm ./src
error: failed to link /tmp/tinygo705502756/main: none of these commands were found in your $PATH: wasm-ld-10 wasm-ld

➜ tinygo version
tinygo version 0.13.1 linux/amd64 (using go version go1.14.2 and LLVM version 10.0.0)
```

is it worth opening an issue?

PR #323 (2019-05-05): main: add fallback mechanism for LLVM commands
On Debian, all LLVM commands have a version suffix (clang-8, ld.lld-8,
wasm-ld-8, etc.). However. Most other distributions only provide a
version prefix for Clang and not for all the other commands.

This commit fixes the issue by trying the command with the version
suffix first and falling back to one without if needed.

fixes #305.