tinygo-org/tinygo #309: Stack overflow when generating wasm with strings.Builder

justinclift, 2019-04-27:
Hit an interesting stack overflow error when experimenting with TinyGo (dev branch, commit 9a3d0683b34f9f030e1d0323bea849d56054cdec) again just now:

```
$ tinygo build -o wasm.wasm -opt=1 -target wasm wasm.go
runtime: goroutine stack exceeds 1000000000-byte limit
fatal error: stack overflow

runtime stack:
runtime.throw(0x3406dd4, 0xe)
        /usr/local/go/src/runtime/panic.go:617 +0x72
runtime.newstack()
        /usr/local/go/src/runtime/stack.go:1041 +0x6f0
runtime.morestack()
        /usr/local/go/src/runtime/asm_amd64.s:429 +0x8f

goroutine 1 [running]:
bytes.(*Buffer).WriteString(0xc00555b8f0, 0xc0000f6671, 0x7, 0x0, 0x0, 0x8)
        /usr/local/go/src/bytes/buffer.go:183 +0x117 fp=0xc024000370 sp=0xc024000368 pc=0x8f1907
go/types.writePackage(0xc00555b8f0, 0xc0014b5b80, 0x0)
        /usr/local/go/src/go/types/object.go:443 +0x64 fp=0xc0240003b0 sp=0xc024000370 pc=0xa15eb4
go/types.writeType(0xc00555b8f0, 0x34881e0, 0xc0013094a0, 0x0, 0xc024000528, 0x1, 0x8)
        /usr/local/go/src/go/types/typestring.go:233 +0xbda fp=0xc0240004e8 sp=0xc0240003b0 pc=0xa2a66a
go/types.WriteType(...)
        /usr/local/go/src/go/types/typestring.go:75
go/types.TypeString(0x34881e0, 0xc0013094a0, 0x0, 0x0, 0xc0004c4000)
        /usr/local/go/src/go/types/typestring.go:67 +0xb1 fp=0xc0240005b8 sp=0xc0240004e8 pc=0xa299f1
go/types.(*Named).String(...)
        /usr/local/go/src/go/types/type.go:490
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x34881e0, 0xc0013094a0, 0xc00555b8c0, 0x30)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/interface.go:96 +0x1010 fp=0xc024000768 sp=0xc0240005b8 pc=0xb79aa0
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x3488220, 0xc000793950, 0x2, 0xc00555e2e0)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/interface.go:158 +0xe54 fp=0xc024000918 sp=0xc024000768 pc=0xb798e4

...

github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x34881e0, 0xc0013094a0, 0xc00555aff0, 0x30)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/interface.go:178 +0x54a fp=0xc02400a608 sp=0xc02400a458 pc=0xb78fda
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x3488220, 0xc000793950, 0x2, 0xc005555d00)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/interface.go:158 +0xe54 fp=0xc02400a7b8 sp=0xc02400a608 pc=0xb798e4

goroutine 5 [syscall]:
os/signal.signal_recv(0x0)
        /usr/local/go/src/runtime/sigqueue.go:139 +0x9c
os/signal.loop()
        /usr/local/go/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
        /usr/local/go/src/os/signal/signal_unix.go:29 +0x41
$
```

The code which caused it is:

```
	var colStr strings.Builder
	colStr.WriteString("rgb(")
	colStr.WriteString(strconv.Itoa(colR))
	colStr.WriteString(", ")
	colStr.WriteString(strconv.Itoa(colG))
	colStr.WriteString(", ")
	colStr.WriteString(strconv.Itoa(colB))
	colStr.WriteString(")")
```

The full code (I'll probably leave it around in a branch, just for this for a while) is here if that helps:

https://github.com/justinclift/tinygo_canvas_test1/blob/stackoverflow_1/wasm.go

aykevl, 2019-05-03:
I managed to reduce the crash to this sample:

```go
package main

type Builder struct {
	addr *Builder
}

func main() {
	var itf interface{}
	itf = Builder{}
	_ = itf
}
```

aykevl, 2019-05-03:
This will probably get fixed when I implement support for named types in reflect.

justinclift, 2019-05-04:
Cool. :smile:

aykevl, 2019-08-06:
This bug has been fixed in the dev branch.

closed 2019-08-06 by aykevl

SuryaprakashBS, 2019-09-05:
still I was faced this issue .

PR #474 (2019-08-05): Improve interface and reflect support
Multiple commits to add more support for reflect:

1.  compiler: fix crash with linked lists in interfaces
    Also, it prepares for some other reflect-related changes that should make it easier to add support for named types (etc.) in the future.
   Together with #472, this should make the `"image/png"` package importable.
2. reflect: add support for named types

fixes #309