tinygo-org/tinygo #310: wasm: SIGSEV from time.Sleep()

justinclift, 2019-04-27:
Hit an interesting stack overflow error when experimenting with TinyGo (dev branch, commit 9a3d0683b34f9f030e1d0323bea849d56054cdec) again just now:

```
$ tinygo build -o wasm.wasm -no-debug -target wasm wasm.go
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x30bb930]

runtime stack:
runtime.throw(0x3413192, 0x2a)
        /usr/local/go/src/runtime/panic.go:617 +0x72
runtime.sigpanic()
        /usr/local/go/src/runtime/signal_unix.go:374 +0x4a9

goroutine 1 [syscall]:
runtime.cgocall(0xbce1d0, 0xc001b20fd8, 0x3401a00)
        /usr/local/go/src/runtime/cgocall.go:128 +0x5b fp=0xc001b20fa8 sp=0xc001b20f70 pc=0x87e65b
github.com/tinygo-org/tinygo/vendor/tinygo.org/x/go-llvm._Cfunc_LLVMTypeOf(0x0, 0x0)
        _cgo_gotypes.go:9016 +0x4a fp=0xc001b20fd8 sp=0xc001b20fa8 pc=0xa4f66a
github.com/tinygo-org/tinygo/vendor/tinygo.org/x/go-llvm.Value.Type.func1(0x0, 0x200000003)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/vendor/tinygo.org/x/go-llvm/ir.go:679 +0x56 fp=0xc001b21010 sp=0xc001b20fd8 pc=0xa642f6
github.com/tinygo-org/tinygo/vendor/tinygo.org/x/go-llvm.Value.Type(0x0, 0x8bca1c)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/vendor/tinygo.org/x/go-llvm/ir.go:679 +0x2b fp=0xc001b21030 sp=0xc001b21010 pc=0xa52f4b
github.com/tinygo-org/tinygo/compiler.(*Compiler).expandFormalParam(0xc0000ae820, 0x0, 0x3, 0xc0027b05e0, 0x8bcc52)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/calls.go:60 +0x2f fp=0xc001b21070 sp=0xc001b21030 pc=0xb49def
github.com/tinygo-org/tinygo/compiler.(*Compiler).createCall(0xc0000ae820, 0x7f407c00c998, 0xc0027b05c0, 0x3, 0x4, 0x0, 0x0, 0x4)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/calls.go:34 +0x116 fp=0xc001b21100 sp=0xc001b21070 pc=0xb49b76
github.com/tinygo-org/tinygo/compiler.(*Compiler).createRuntimeCall(0xc0000ae820, 0x3405c7f, 0xc, 0xc00245bd60, 0x2, 0x2, 0x0, 0x0, 0x532ca70)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/calls.go:27 +0x162 fp=0xc001b21170 sp=0xc001b21100 pc=0xb49752
github.com/tinygo-org/tinygo/compiler.(*Compiler).markAsyncFunctions(0xc0000ae820, 0xc001b21558, 0xc07c039970, 0x7c0399700087ec65)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/goroutine-lowering.go:382 +0x20e8 fp=0xc001b214f0 sp=0xc001b21170 pc=0xb6fd88
github.com/tinygo-org/tinygo/compiler.(*Compiler).LowerGoroutines(0xc0000ae820, 0x7f407c00aa10, 0x7f407c0379f8)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/goroutine-lowering.go:109 +0x40 fp=0xc001b215a8 sp=0xc001b214f0 pc=0xb6d720
github.com/tinygo-org/tinygo/compiler.(*Compiler).Optimize(0xc0000ae820, 0x2, 0x2, 0x5, 0x0, 0x0)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/compiler/optimizer.go:77 +0x512 fp=0xc001b216b0 sp=0xc001b215a8 pc=0xb7e7b2
main.Compile(0x7ffd03893d3f, 0x7, 0x7ffd03893d1e, 0x9, 0xc0000c82c0, 0xc001b21f00, 0xc001b21d88, 0x0, 0x0)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/main.go:144 +0x2e9c fp=0xc001b21d40 sp=0xc001b216b0 pc=0xbbad9c
main.Build(0x7ffd03893d3f, 0x7, 0x7ffd03893d1e, 0x9, 0x7ffd03893d3a, 0x4, 0xc00009df00, 0xc0000645e0, 0xffffffffffffffff)
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/main.go:296 +0xe2 fp=0xc001b21db0 sp=0xc001b21d40 pc=0xbbb502
main.main()
        /home/jc/git_repos/src/github.com/tinygo-org/tinygo/main.go:589 +0xc74 fp=0xc001b21f98 sp=0xc001b21db0 pc=0xbbd194
runtime.main()
        /usr/local/go/src/runtime/proc.go:200 +0x20c fp=0xc001b21fe0 sp=0xc001b21f98 pc=0x8a8edc
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1337 +0x1 fp=0xc001b21fe8 sp=0xc001b21fe0 pc=0x8d35a1

goroutine 5 [syscall]:
os/signal.signal_recv(0x0)
        /usr/local/go/src/runtime/sigqueue.go:139 +0x9c
os/signal.loop()
        /usr/local/go/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
        /usr/local/go/src/os/signal/signal_unix.go:29 +0x41
```

The code which caused this sigsev is:

```
time.Sleep(time.Second)
```

Full code is here, in case it helps:

https://github.com/justinclift/tinygo_canvas_test1/blob/time_sleep_sigsev_1/wasm.go#L63

aykevl, 2019-04-28:
It's unlikely the bare `time.Sleep` is causing the issue, as this is tested in the standard compiler tests.
Can you try narrowing it further down to the smallest possible code that reproduces this issue?

aykevl, 2019-04-28:
On second thought, it's probably because you're trying to call `time.Sleep` in an exported function. That is not supported and the compiler should show an error about that instead of crashing.

The reason this is unsupported is because `time.Sleep` blocks and JavaScript cannot normally block.

justinclift, 2019-04-28:
No worries.  Thanks for taking the time to investigate @aykevl. :smile:

closed 2019-04-28 by justinclift