tinygo-org/tinygo #315: error: async function (*mydriver.SPIBus).ImpFunction incorrectly used in bitcast, expected runtime.makeGoroutine

conejoninja, 2019-04-28:
I don't know how to describe this issue, feel free to modify the title or ask more questions if not clear enough. I got this error when adding _time.Sleep_ to a function (ImpFunction), I call this function through its interface (Buser).

This is a driver I'm working on (simplified):
```go
package mydriver

import (
	"machine"
)

type Device struct {
	bus   Buser
}

type I2CBus struct {
	wire machine.I2C
}

type SPIBus struct {
	wire machine.SPI
}

type Buser interface {
	ImpFunction()
}

func NewI2C(bus machine.I2C) Device {
	return Device{
		bus: &I2CBus{
			wire: bus,
		},
	}
}

func NewSPI(bus machine.SPI) Device {
	return Device{
		bus: &SPIBus{
			wire: bus,
		},
	}
}

func (d *Device) SomeFunction() {
	d.bus.Function()
}
func (b *I2CBus) ImpFunction() {
	println("I2C func")
}

func (b *SPIBus) ImpFunction() {
	println("SPI func")
	time.Sleep(1 * time.Millisecond) // adding this function makes the error
}

```

This is the main/example file that call the driver (very simple):
```go
package main

import (
	"image/color"
	"machine"
	"time"

	"github.com/tinygo-org/drivers/mydriver"
)

func main() {
	machine.SPI0.Configure(machine.SPIConfig{
		Frequency: 8000000,
	})
	display := mydriver.NewSPI(machine.SPI0)
	display.SomeFunction()
}
```

If I remove the time.Sleep function, everything works fine and it outputs _"SPI func"_ or _"I2C func"_ as expected.

aykevl, 2019-04-28:
I managed to narrow this down to this:

```go
package main

import (
	"time"
)

type Buser interface {
	ImpFunction()
}

type SPIBus struct {
}

func (b *SPIBus) ImpFunction() {
	time.Sleep(1 * time.Millisecond) // adding this function makes the error
}

func main() {
	var bus Buser
	bus = &SPIBus{}
	bus.ImpFunction()
}
```

aykevl, 2019-04-29:
@conejoninja #316 should fix this issue, can you test it?

conejoninja, 2019-04-29:
Yes, #316 fixes this

deadprogram, 2019-05-05:
This was merged, so closing.

closed 2019-05-05 by deadprogram