tinygo-org/tinygo #328: Arduino blinky1 with Go 1.12 error

torntrousers, 2019-05-05:
Trying TinyGo 0.5.0 with the blinky1 example on an Arduino and it fails:
```
tinygo flash -target=arduino examples/blinky1
error: cannot find package "unsafe" in any of:
	/usr/local/Cellar/go/1.11.1/libexec/src/unsafe (from $GOROOT)
	/Users/anthonyelder/go/src/unsafe (from $GOPATH)
```
 I don't have Go 1.11.1, only 1.12.1
Is there some simple change I can make somewhere to get it to use Go 1.12.1?

justinclift, 2019-05-05:
Hmmm, it seems like it's looking for the "unsafe" package in your main Go 1.12.1 installation.

On my Linux desktop, where I've downloaded and installed go to `/usr/local/go`, the `unsafe` package is in `/usr/local/go/src/unsafe`.  eg on my system it should find it if either `GOPATH` or `GOROOT` is set to `/usr/local/go`.

Maybe try setting `GOPATH` to `/usr/local/opt/go`, then try again running `tinygo` again?

justinclift, 2019-05-05:
Oh, best to confirm first that `/usr/local/opt/go` does indeed contain your Go install from Homebrew. :smile:

torntrousers, 2019-05-05:
I asked about this in the slack channel [here](https://gophers.slack.com/archives/CDJD3SUP6/p1557052499473400) and ended up installing gvm to use multiple Go versions and just using Go 1.11.1.

closed 2019-05-05 by torntrousers

justinclift, 2019-05-05:
No worries. :smile:

As a data point, Go 1.11.1 is a bit old.  Go 1.11.9 is out, as is Go 1.12.4.  Not sure if that'll make any difference though. :wink:

torntrousers, 2019-05-05:
I don't know where the Go 1.11.1 was coming from, some old env setup on my machine maybe. But now with gvm I've also installed 1.12.4 and that seems to work fine with TinyGo too. 
Thanks for everyones help.

aykevl, 2019-05-05:
For reference, this should be fixed with #330. The cause of this issue was that GOROOT was picked during build and TinyGo on macOS is built with an older version of Go. The fix is to make GOROOT configurable and let it check the environment first before picking a GOROOT.

directionless, 2019-05-07:
FWIW I fixed this by setting `GOROOT` to what go already knows it's set to eg:
```
export GOROOT=$(go env GOROOT)
```

aykevl, 2019-05-07:
...of course, after reading the docs it indeed picks up GOROOT from the environment if set. That's a simple solution. However, #330 should still make things more reliable and would (it there are no bugs) have avoided having to set GOROOT manually.