tinygo-org/tinygo #340: Incorrect IR code generation for runtime.scheduler

justinclift, 2019-05-10:
The IR generated by TinyGo (dev branch 4c8c048c49304bc8ef766177fa404af6bb63e170) for wasm is (at least sometimes) crashing clang (both 8, and 9 `master` head).

Using bugpoint as suggested by @aykevl , the generated output simplifies down to this:

```
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "main.go"
target datalayout = "e-m:e-p:32:32-i64:64-n32:64-S128"
target triple = "wasm32-unknown-unknown-wasm"

define dso_local void @runtime.scheduler() unnamed_addr {
entry:
  br label %for.body

for.body:                                         ; preds = %for.body, %entry
  call void @llvm.coro.resume(i8* undef)
  br label %for.body
}

declare void @llvm.coro.resume(i8*)

!llvm.module.flags = !{!0}

!0 = !{i32 1, !"Debug Info Version", i32 3}
```

The error from clang is:

```
$ clang --compile -v --target=wasm32 -o /dev/null bugpoint-reduced-simplified.bc
clang version 9.0.0 (https://git.llvm.org/git/clang.git d8066f9602863a39f208a52ca3be977c46a5d0f0) (https://git.llvm.org/git/llvm.git b55344a7434712dd066a7e64d2c9691ebd73503e)
Target: wasm32
Thread model: posix
InstalledDir: /opt/llvm9-dbg/bin
 "/opt/llvm9-dbg/bin/clang-9" -cc1 -triple wasm32 -emit-obj -mrelax-all -disable-free -main-file-name bugpoint-reduced-simplified.bc -mrelocation-model static -mthread-model posix -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu generic -fvisibility hidden -dwarf-column-info -debugger-tuning=gdb -momit-leaf-frame-pointer -v -coverage-notes-file /dev/null.gcno -resource-dir /opt/llvm9-dbg/lib/clang/9.0.0 -fdebug-compilation-dir /home/jc/git_repos/src/github.com/tinygo-org/tinygo/src/examples/wasm/main -ferror-limit 19 -fmessage-length 187 -fobjc-runtime=gnustep -fno-common -fdiagnostics-show-option -fcolor-diagnostics -o /dev/null -x ir bugpoint-reduced-simplified.bc
clang -cc1 version 9.0.0 based upon LLVM 9.0.0svn default target x86_64-unknown-linux-gnu
warning: overriding the module target triple with wasm32 [-Woverride-module]
fatal error: error in backend: Cannot select: intrinsic %llvm.coro.resume
clang-9: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 9.0.0 (https://git.llvm.org/git/clang.git d8066f9602863a39f208a52ca3be977c46a5d0f0) (https://git.llvm.org/git/llvm.git b55344a7434712dd066a7e64d2c9691ebd73503e)
Target: wasm32
Thread model: posix
InstalledDir: /opt/llvm9-dbg/bin
clang-9: note: diagnostic msg: PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace, preprocessed source, and associated run script.
clang-9: note: diagnostic msg: Error generating preprocessed source(s) - no preprocessable inputs.
```

justinclift, 2019-05-10:
If it's useful, the full output from clang on the non-reduced file:

```
$ clang --compile -v --target=wasm32-unknown-unknown-wasm -o /dev/null main.ll
clang version 9.0.0 (https://git.llvm.org/git/clang.git d8066f9602863a39f208a52ca3be977c46a5d0f0) (https://git.llvm.org/git/llvm.git b55344a7434712dd066a7e64d2c9691ebd73503e)
Target: wasm32-unknown-unknown-wasm
Thread model: posix
InstalledDir: /opt/llvm9-dbg/bin
 "/opt/llvm9-dbg/bin/clang-9" -cc1 -triple wasm32-unknown-unknown-wasm -emit-obj -mrelax-all -disable-free -main-file-name main.ll -mrelocation-model static -mthread-model posix -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu generic -fvisibility hidden -dwarf-column-info -debugger-tuning=gdb -momit-leaf-frame-pointer -v -coverage-notes-file /dev/null.gcno -resource-dir /opt/llvm9-dbg/lib/clang/9.0.0 -fdebug-compilation-dir /home/jc/git_repos/src/github.com/tinygo-org/tinygo/src/examples/wasm/main -ferror-limit 19 -fmessage-length 187 -fobjc-runtime=gnustep -fno-common -fdiagnostics-show-option -fcolor-diagnostics -o /dev/null -x ir main.ll
clang -cc1 version 9.0.0 based upon LLVM 9.0.0svn default target x86_64-unknown-linux-gnu
PromoteIntegerResult #0: t4: i1,ch = llvm.coro.done t0, TargetConstant:i32<17>, t3, scheduler.go:135:11

Do not know how to promote this operator!
UNREACHABLE executed at /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/LegalizeIntegerTypes.cpp:54!
Stack dump:
0.      Program arguments: /opt/llvm9-dbg/bin/clang-9 -cc1 -triple wasm32-unknown-unknown-wasm -emit-obj -mrelax-all -disable-free -main-file-name main.ll -mrelocation-model static -mthread-model posix -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu generic -fvisibility hidden -dwarf-column-info -debugger-tuning=gdb -momit-leaf-frame-pointer -v -coverage-notes-file /dev/null.gcno -resource-dir /opt/llvm9-dbg/lib/clang/9.0.0 -fdebug-compilation-dir /home/jc/git_repos/src/github.com/tinygo-org/tinygo/src/examples/wasm/main -ferror-limit 19 -fmessage-length 187 -fobjc-runtime=gnustep -fno-common -fdiagnostics-show-option -fcolor-diagnostics -o /dev/null -x ir main.ll 
1.      Code generation
2.      Running pass 'Function Pass Manager' on module 'main.ll'.
3.      Running pass 'WebAssembly Instruction Selection' on function '@runtime.runqueuePushBack'
 #0 0x0000000005ce0699 llvm::sys::PrintStackTrace(llvm::raw_ostream&) /home/jc/local/source/llvm/lib/Support/Unix/Signals.inc:494:11
 #1 0x0000000005ce0849 PrintStackTraceSignalHandler(void*) /home/jc/local/source/llvm/lib/Support/Unix/Signals.inc:558:1
 #2 0x0000000005cdf1d6 llvm::sys::RunSignalHandlers() /home/jc/local/source/llvm/lib/Support/Signals.cpp:67:5
 #3 0x0000000005ce0e7b SignalHandler(int) /home/jc/local/source/llvm/lib/Support/Unix/Signals.inc:357:1
 #4 0x00007f61172625d0 __restore_rt (/lib64/libpthread.so.0+0xf5d0)
 #5 0x00007f611607b207 raise (/lib64/libc.so.6+0x36207)
 #6 0x00007f611607c8f8 abort (/lib64/libc.so.6+0x378f8)
 #7 0x0000000005c47fe0 llvm::install_out_of_memory_new_handler() /home/jc/local/source/llvm/lib/Support/ErrorHandling.cpp:192:0
 #8 0x00000000072125b0 llvm::DAGTypeLegalizer::PromoteIntegerResult(llvm::SDNode*, unsigned int) /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/LegalizeIntegerTypes.cpp:55:59
 #9 0x00000000071c0e2c llvm::DAGTypeLegalizer::run() /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/LegalizeTypes.cpp:253:17
#10 0x00000000071c622b llvm::SelectionDAG::LegalizeTypes() /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/LegalizeTypes.cpp:1114:3
#11 0x00000000071a5786 llvm::SelectionDAGISel::CodeGenAndEmitDAG() /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:805:13
#12 0x00000000071a5113 llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Instruction, false, false, void>, false, true>, llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Instruction, false, false, void>, false, true>, bool&) /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:699:1
#13 0x00000000071a4656 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:1754:15
#14 0x00000000071a2326 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/jc/local/source/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:496:3
#15 0x000000000499a28b (anonymous namespace)::WebAssemblyDAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/jc/local/source/llvm/lib/Target/WebAssembly/WebAssemblyISelDAGToDAG.cpp:56:5
#16 0x0000000004f81b97 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) /home/jc/local/source/llvm/lib/CodeGen/MachineFunctionPass.cpp:73:8                                       
#17 0x0000000005487159 llvm::FPPassManager::runOnFunction(llvm::Function&) /home/jc/local/source/llvm/lib/IR/LegacyPassManager.cpp:1648:23                                                 
#18 0x000000000548759f llvm::FPPassManager::runOnModule(llvm::Module&) /home/jc/local/source/llvm/lib/IR/LegacyPassManager.cpp:1685:16                                                     
#19 0x0000000005487d05 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) /home/jc/local/source/llvm/lib/IR/LegacyPassManager.cpp:1752:23                                    
[... cut ...]
clang version 9.0.0 (https://git.llvm.org/git/clang.git d8066f9602863a39f208a52ca3be977c46a5d0f0) (https://git.llvm.org/git/llvm.git b55344a7434712dd066a7e64d2c9691ebd73503e)
Target: wasm32-unknown-unknown-wasm
Thread model: posix
InstalledDir: /opt/llvm9-dbg/bin
clang-9: note: diagnostic msg: PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace, preprocessed source, and associated run script.
clang-9: note: diagnostic msg: Error generating preprocessed source(s) - no preprocessable inputs.
```

justinclift, 2019-05-10:
The IR itself is generated from TinyGo using `src/examples/wasm/main/main.go` as the input source:

```
$ tinygo build -printir -target wasm -o /dev/null src/examples/wasm/main/main.go > main.ll
```

justinclift, 2019-05-10:
As a data point for future reference, the bugpoint command used to reduce the test case was:

```
$ bugpoint main.ll -llc-safe -safe-tool-args -mtriple=wasm32-unknown-unknown-wasm
```

aykevl, 2019-05-11:
I think the issue here is that you did not run the required LLVM coroutine lowering passes. These are CoroEarly, CoroSplit, CoroElide and CoroCleanup. These passes are required for correct functioning of coroutines in LLVM. You can run them manually with the `opt` tool (`-coro-early` etc). They likely need a few more optimization passes to be enabled or they will crash.

More background:
https://llvm.org/docs/Coroutines.html
https://youtu.be/Ztr8QvMhqmQ (long watch but goes into a lot of detail, possibly a bit outdated)

aykevl, 2019-05-11:
For now, I wouldn't worry too much about this. If you want to manually tweak the IR and don't want to do anything with coroutines, you can probably just do a `-globalopt` pass to discard unused functions.

justinclift, 2019-05-11:
Thanks @aykevl, makes sense. :smile:

aykevl, 2019-05-13:
I think there's nothing to fix here. TinyGo is working as intended. The IR generated here is not final and should generally not be fed directly to Clang or llc (in fact, it contains some incomplete code that is lowered in a later stage during optimization). If you want to compile IR, emit code using `tinygo build -o test.ll -opt=1` (or even `-opt=0`).

closed 2019-05-13 by aykevl

justinclift, 2019-05-14:
Agreed.  Was trying out various `opt` and `tinygo build -o test.ll` ll / bitcode generation stuff yesterday, but it still kept crashing clang.

Turns out it's a clang bug when `-o /dev/null` is used with wasm generation.  Aargh. :wink:

&nbsp; &nbsp; https://bugs.llvm.org/show_bug.cgi?id=41857