tinygo-org/tinygo #343: I2C reads losing a byte

torntrousers, 2019-05-10:
I've a bug somewhere and I'm starting ti wonder if its in TinyGo. The problem is with reading data from a u-blox GPS module over I2C and it seems to lose a byte after each read. 
This is some code that recreates the problem:
```
func main() {
	println("Test1")
	machine.I2C0.Configure(machine.I2CConfig{})
	var addr uint16 = 0x42
	// var data []byte = make([]byte, 10)
	var data [10]byte
	for {
		// time.Sleep(17 * time.Millisecond)
		machine.I2C0.Tx(addr, []byte{0xFF}, data[:])
		print("[")
		print(string(data[:]))
		println("]")
	}
}
```
and that outputs things like:
```
[,,,,*7A
$]
[NGSA,A,1,,]
```
which has lost a 'G' between the '$' and 'NGSA'

It seems to happen after each Tx call the next byte from the slave is lost.

Any ideas on how I could de bug this?

aykevl, 2019-05-11:
What target is this for? Such problems are highly chip specific.

torntrousers, 2019-05-11:
Running on a Microbit

torntrousers, 2019-05-11:
I think I found it, moving this line https://github.com/tinygo-org/tinygo/blob/master/src/machine/machine_nrf.go#L238 up to be just before line 235 seem to fix the problem. Doing some more testing...

torntrousers, 2019-05-11:
Running for a couple of hours now without missing a beat so it does look like #344 fixes this

deadprogram, 2019-05-11:
Did you happen to try your change with any other I2C devices?

aykevl, 2019-05-11:
This could explain why I had some weird issues with I2C sometimes, with data that was off by one byte sometimes (but not always).

torntrousers, 2019-05-11:
I've tried now with a couple of other I2C devices. The SHT31 from the open PR works with and without the fix. The BMP180 though doesn't work with this fix on. Hmm.

torntrousers, 2019-05-11:
Seems to be some timing thing - if I put a println at the start of machine_nrf.go readLastByte then the BMP180 example works with this fix.

torntrousers, 2019-05-11:
Fixed by moving to after the EVENTS_RXDREADY loop. So now it works with BMP180, SHT31, and the u-blox MAX-8.

torntrousers, 2019-05-11:
Bugger, no the u-blox still misses a byte sometimes like this.

torntrousers, 2019-05-12:
I can't figure out how to get this working with both the BMP180 and the u-blox GPS. Annoying. Any ideas anyone?

deadprogram, 2019-05-12:
Was doing some searches and came up with this: https://github.com/sandeepmistry/arduino-nRF5/pull/137

Seems like trying to use the "short" might be the way to go here.

torntrousers, 2019-05-12:
I've hacked around using that SHORTS/TASKS_RESUME without really understanding what I'm doing and amazingly it seems to work. Need to go but now but will show that code in a PR later on today.

deadprogram, 2019-05-13:
This has been merged into `dev` branch, so now closing. Thanks @torntrousers for the fix.

closed 2019-05-13 by deadprogram