tinygo-org/tinygo #351: error: verification error after IR construction

deadprogram, 2019-05-14:
When building the https://github.com/tinygo-org/drivers/tree/master/examples/espat/espconsole example as part of the build in the smoke test, the following error is now appearing:

```shell
$ tinygo build -size short -o test.elf -target=itsybitsy-m0 ./examples/espat/espconsole/main.go
conflicting debug info for argument
  call void @llvm.dbg.value(metadata i8* %0, metadata !30275, metadata !DIExpression(DW_OP_LLVM_fragment, 0, 32)), !dbg !30291
!30280 = !DILocalVariable(name: "s", arg: 1, scope: !30278, file: !29386, line: 171, type: !127)
!30275 = !DILocalVariable(name: "s", arg: 1, scope: !30270, file: !30271, line: 150, type: !127)
conflicting debug info for argument
  call void @llvm.dbg.value(metadata i8 %2, metadata !30276, metadata !DIExpression()), !dbg !30291
!30281 = !DILocalVariable(name: "c", arg: 2, scope: !30278, file: !29386, line: 171, type: !162)
!30276 = !DILocalVariable(name: "c", arg: 2, scope: !30270, file: !30271, line: 150, type: !162)
error: verification error after IR construction
```

deadprogram, 2019-05-14:
Verified that I can reproduce same issue on my machine.

deadprogram, 2019-05-14:
From using `git bisect` it would appear that commit 371c468e8efcc6c415c4e955e69fc6072698f578 is what introduced this error:

```shell
$ git bisect bad 3c2639ad5520434a37488cedf3917630e7266964
Bisecting: 0 revisions left to test after this (roughly 1 step)
[371c468e8efcc6c415c4e955e69fc6072698f578] compiler: add debug info for function arguments
```

aykevl, 2019-05-14:
Thanks, will have to investigate. As a workaround, you can add `-no-debug`.

aykevl, 2019-05-14:
Thinking of it, perhaps we should also run the driver smoke tests as part of the compiler CI? That would catch bugs like these.

deadprogram, 2019-05-14:
I was thinking about exactly that. Seems like a very good idea.

aykevl, 2019-05-14:
... really weird, a function is being defined twice. And the first one is incorrect. This is probably an already existing bug that only surfaces after adding debug info.

aykevl, 2019-05-14:
This is the cause: https://github.com/golang/go/commit/ad4a58e31501bce5de2aad90a620eaecdc1eecb8

aykevl, 2019-05-14:
See #352.

deadprogram, 2019-05-24:
Verified locally:

```shell
$ tinygo build -size short -o ./build/test.elf -target=itsybitsy-m0 ./examples/espat/espconsole/main.go

   code    data     bss |   flash     ram
  23184    1420    4468 |   24604    5888
```

Looks good!

PR #352 (2019-05-24): compiler,runtime: fix multiple definitions of a single function
`strings.IndexByte` used to be implemented in the runtime. Now it is implemented using a direct call to `internal/bytealg.IndexByte`. Fixes #351.

The real problem is of course https://github.com/golang/go/issues/31330.
closed 2019-05-24 by deadprogram