tinygo-org/tinygo #363: panic: branch was not true or false

deadprogram, 2019-05-20:
I am working on the refactoring of the NRF machine/runtime to use the volatile package.

Here is the WIP branch https://github.com/tinygo-org/tinygo/tree/feature/nrf-refactor-volatile

When I compile the `microbit-blink` example, I receive the following panic:

```shell
$ tinygo build -size short -o test.elf -target=microbit examples/microbit-blink
panic: branch was not true or false

goroutine 1 [running]:
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc001409330, 0x7f7b4804c260, 0x7f7b4804b9c0, 0xc0011a2e74, 0xc, 0x0, 0x0, 0xc000fd02d8, 0x1, 0x1, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/frame.go:473 +0x89f9
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc0013f2a50, 0x7f7b4800a878, 0xc000cc8a00, 0x4, 0x4, 0xc000fd51d0, 0x7, 0xc0011a2e74, 0xc, 0x8c5a00, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/interp.go:104 +0x208
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc001409e50, 0x7f7b48047250, 0x7f7b48046e30, 0xc000fd5f78, 0x8, 0x0, 0x0, 0xc000fd02c0, 0x1, 0x1, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/frame.go:414 +0x4eae
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc0013f2a50, 0x7f7b4800a688, 0xc000cb4e00, 0x6, 0x8, 0xc000fd51d0, 0x7, 0xc000fd5f78, 0x8, 0x8c5a00, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/interp.go:104 +0x208
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc00140a970, 0x7f7b4813b4f0, 0x7f7b4813b0d0, 0x8359b0, 0x4, 0x0, 0x0, 0xc000fd0258, 0x1, 0x1, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/frame.go:414 +0x4eae
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc0013f2a50, 0x7f7b48019d78, 0xc000f8c8e0, 0x2, 0x2, 0xc000fd51d0, 0x7, 0x8359b0, 0x4, 0x8c5a00, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/interp.go:104 +0x208
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc00140b490, 0x7f7b48135330, 0x0, 0x0, 0x0, 0xc04826a8e0, 0x10, 0x10, 0xc000f8fa20, 0x5e8e31, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/frame.go:414 +0x4eae
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc0013f2a50, 0x7f7b48019b98, 0xc00140b5d0, 0x2, 0x2, 0xc000fd51d0, 0x7, 0x0, 0x0, 0x0, ...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/interp.go:104 +0x208
github.com/tinygo-org/tinygo/interp.(*Eval).Function(...)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/interp.go:87
github.com/tinygo-org/tinygo/interp.Run(0x1e6e150, 0x1e6c6c0, 0x0, 0x0, 0x0)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/interp/interp.go:74 +0x672
main.Compile(0x7ffea3b908ff, 0x17, 0x7ffea3b908e5, 0x8, 0xc00013a160, 0xc00140bf00, 0xc00140bd88, 0x0, 0x0)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/main.go:120 +0x8cc
main.Build(0x7ffea3b908ff, 0x17, 0x7ffea3b908e5, 0x8, 0x7ffea3b908f6, 0x8, 0xc00008bf00, 0xc0000fc1b0, 0xc0000fe1a0)
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/main.go:311 +0xe2
main.main()
        /home/ron/.gvm/pkgsets/go1.12/global/src/github.com/tinygo-org/tinygo/main.go:604 +0xcc0
```

aykevl, 2019-05-21:
Fix in #366.

deadprogram, 2019-05-21:
New problem, probably related:

```shell
$ tinygo build -size short -o test.elf -target=pca10040   examples/blinky1
   code    data     bss |   flash     ram
   2060       4    2212 |    2064    2216
$ tinygo build -size short -o test.elf -target=pca10056   examples/blinky1
error: interp: branch on a non-const-propagated constant expression
```

aykevl, 2019-05-24:
Should be fixed in #372.

deadprogram, 2019-05-24:
Confirm that it fixes the above issue:

```shell
$ tinygo build -size short -o test.elf -target=pca10040   examples/blinky1
   code    data     bss |   flash     ram
   2060       4    2212 |    2064    2216
```

Thanks!

closed 2019-05-24 by deadprogram

aykevl, 2019-05-24:
You mean this?
```
$ go install; and tinygo build -size short -o test.elf -target=pca10056   examples/blinky1
   code    data     bss |   flash     ram
   1804       4    4260 |    1808    4264
```

deadprogram, 2019-05-24:
Also that one.