tinygo-org/tinygo #364: strconv.ParseInt panic when arg not an int

torntrousers, 2019-05-20:
Doing `var x, e = strconv.ParseInt("12a", 10, 32)` gives `panic: unimplemented: interface equality`, whereas `var x, e = strconv.ParseFloat("1.2a", 32)` works correctly and e gets an Error `strconv.ParseFloat: parsing "1.2a": invalid syntax`

The panic comes from [here](https://github.com/tinygo-org/tinygo/blob/master/src/runtime/interface.go#L26) -  I added some debug prints just before the panic:
`
	fmt.Println(x.typecode)
	fmt.Println(y.typecode)
	fmt.Println(x.typecode != y.typecode)
`
and they output:
`
7141
7141
false
`  
so I don't know why they are not equal.

Any ideas on how to debug and/or fix this?

aykevl, 2019-05-20:
The function `runtime.interfaceEqual` is a compiler intrinsic that is inserted when you try to compare two values of interface type. [For example](https://play.golang.org/p/Wrvc1QB1H5a):

```go
func isItZero(itf interface{}) bool {
	return itf == 0 // implemented as call to runtime.interfaceEqual
}
```

It has two fast paths for common cases but to be generic there must be some sort of run-time type information. It cannot simply compare the value:

* two NaN float values, while sharing the same bit pattern, are never equal
* two strings will be stored in pointers so have a different bit pattern, but can still be equal

What you see in the output is that the typecodes are equal. That means the values _may be_ equal, but may also not be equal. Some sort of runtime type information is required to do the compare. This has not been implemented, and is certainly not trivial.

torntrousers, 2019-05-20:
Its interesting that strconv.ParseFloat works ok though - you'd think that they both might use similar code to spot non numeric characters so how does ParseFloat work but not ParseInt?

aykevl, 2019-05-20:
You'd have to look in the standard library for that. You can use `gdb` to debug it, for example:

```
tinygo gdb ./test.go
[...]
Starting program: /tmp/tinygo719036714/main 
f: 3.14
i: 0
i: 1
i: 2
i: 3
i: 4
i: 5
i: 6
i: 7
i: 8
i: 9
panic: asdf

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff7a6e42a in __GI_abort () at abort.c:89
#2  0x0000000000203e28 in ./tmp/strconv.go.main () at tmp/strconv.go:10
#3  0x00000000002031c3 in main () at src/runtime/runtime_unix.go:50
(gdb) quit
```

Also, I cannot reproduce the issue locally with the following code:

```go
package main

import "strconv"

func main() {
	println("f:", strconv.FormatFloat(3.14, 'G', -1, 64))
	for i := int64(0); i < 10; i++ {
		println("i:", strconv.FormatInt(i, 10))
	}
}
```

Are you sure the issue lies in strconv and not somewhere else?

torntrousers, 2019-05-20:
This reproduces for me:
`
package main

import (
	"strconv"
)

func main() {
	strconv.ParseInt("12a", 10, 32)
}
`
It seems a bit hard to get gdb working on a Mac and I've not been able to get it to work yet but will keep trying...

aykevl, 2019-05-20:
Of course, ParseInt instead of FormatInt. I didn't read correctly.
Also, you only hit this error when you pass an incorrect value, it works just fine for properly formatted integers.

You can also use lldb to get a stack trace, which is probably easier to use on macOS:

```
$ tinygo build -o test ./tmp/strconv.go
$ lldb-8 ./test
(lldb) target create "./test"
Current executable set to './test' (x86_64).
(lldb) run
Process 18319 launched: '/home/ayke/src/github.com/tinygo-org/tinygo/test' (x86_64)
panic: unimplemented: interface equality
Process 18319 stopped
* thread #1, name = 'test', stop reason = signal SIGABRT
    frame #0: 0x00007ffff7a6cfff libc.so.6`__GI_raise(sig=6) at raise.c:51
(lldb) bt
* thread #1, name = 'test', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff7a6cfff libc.so.6`__GI_raise(sig=6) at raise.c:51
    frame #1: 0x00007ffff7a6e42a libc.so.6`__GI_abort at abort.c:89
    frame #2: 0x00000000002011b6 test`runtime.interfaceEqual(x=<unavailable>, y=<unavailable>) at interface.go:26:7
    frame #3: 0x00000000002012df test`main [inlined] strconv.ParseInt(s=(ptr = <no value available>, len = 3), base=10, bitSize=32) at atoi.go:177:39
    frame #4: 0x00000000002011f2 test`main [inlined] ./tmp/strconv.go.main at strconv.go:8
    frame #5: 0x00000000002011f2 test`main at runtime_unix.go:50
    frame #6: 0x00007ffff7a5a2e1 libc.so.6`__libc_start_main(main=(test`main [inlined] runtime.initAll at runtime_unix.go:47
test`main at runtime_unix.go:47), argc=1, argv=0x00007fffffffe4f8, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffe4e8) at libc-start.c:291
    frame #7: 0x000000000020102a test`_start + 42
(lldb) 
```

aykevl, 2019-11-26:
This issue should be fixed in the latest dev branch after https://github.com/tinygo-org/tinygo/pull/686 was merged.

closed 2019-11-26 by aykevl