tinygo-org/tinygo #368: Fix compatibility with Go 1.12 stdlib

aykevl, 2019-05-23:
It looks like when TinyGo is compiled with Go 1.11, it does not support the stdlib of Go 1.12. It throws errors like the following:

```
syscall/syscall_aix.go:26:2: F_DUPFD_CLOEXEC redeclared in this block
syscall/net_js.go:44:2: 	other declaration of F_DUPFD_CLOEXEC
```

Either we should start building TinyGo on macOS with Go 1.12, or fix this incompatibility (whatever it may be).

aykevl, 2019-05-24:
It looks like the go/build package hardcodes the list of known GOOS/GOARCH values, so we can't really fix this. This means we'll likely need to start building TinyGo using Go 1.12 on macOS.

deadprogram, 2019-05-24:
macOS users that installed go using Homebrew will be using go 1.12 already, so seems good.

aykevl, 2019-05-24:
But TinyGo is currently built with Go 1.11 which leads to this error.

I think it's something like this:

| | go1.11 stdlib | go1.12 stdlib |
| --- | --- | --- |
| tinygo compiled with go1.11 | OK | FAIL |
| tinygo compiled with go1.12 | OK | OK |

aykevl, 2019-05-27:
This should be fixed with #383.

deadprogram, 2019-05-28:
OK, in that case let's close this issue.

closed 2019-05-28 by deadprogram