tinygo-org/tinygo #379: Zephyr project integration

gedw99, 2019-05-26:
https://docs.zephyrproject.org/latest/

We are using tinygo and it's been a huge help.
Now we are looking into the best way to support the OS and BSP aspects.

I know about the yocto project, but am interested in using tinygo with zephyr to get a complete architecture.

Is this something that others are using I wonder ? Anyone have experience using it with Tinygo ?

deadprogram, 2019-05-28:
Hi @gedw99 integrating with Zephyr would be great. We've looked at it a little alongside a few others.

Most RTOS will require similar changes. For example, all will require some way of scheduling goroutines, probably in separate tasks instead of how it is implemented now.

I'd love to find out more about your current use cases if you do not mind sharing, or you can contact me privately if you prefer that.

gedw99, 2019-06-06:
Hey @deadprogram 
Medical devices is what we are using tinygo for.
So EKG and Glucose Continuous meters. 

We are not at all good a this actually. More server programmers. Tinygo was chosen because everything else is golang :) Poor excuse i know but for now it works for us.

Updatehub has a golang SDK and uses Zephyr and is able to do transaction A/B updates to the iot devices. That gets that huge hurdle sorted. Managing IOT devices so you dont brick them is hard.
https://github.com/UpdateHub
We are collaborating with them on this aspect but also embedded in general since we are not embedded architects.

aykevl, 2019-06-07:
Great to see your interest in using TinyGo for such a serious application!
As Ron mentioned, we don't have support for RTOSes now because it's not really our primary target, but would be interested in supporting them in the future. Challenges here are goroutine starting/scheduling, build tool integration (who manages the build, TinyGo or west/newt/whatever), and heap management (Go needs garbage collection).

> Tinygo was chosen because everything else is golang :)

This is one of the reasons why we're making it. Our hope is that it allows code reuse between projects lowering the cost of development.

gordonklaus, 2019-08-02:
Hi.  Can someone explain to me what the point of Zephyr integration with TinyGo would be?  I came to TinyGo with the impression that it is more of a replacement for Zephyr, but I would love to know in what regards that is not the case.

deadprogram, 2019-08-04:
The idea to me is to be able to use TinyGo as the compiler, and use capabilities in the RTOS, along with interoperating with other services/libs written for that RTOS.

It is true that TinyGo also has a Hardware Abstration Layer (HAL) of its own, but those are primarily related to peripheral i/o such as I2C, SPI, etc. Seems like you could even also use these things together, stitching together code that use the TinyGo HAL alongside services such as OTA updates, etc.

aykevl, 2019-08-04:
Agreed with @deadprogram here. Yes, the runtime and machine packages replace most of the stuff you would normally expect from a RTOS. However, many projects are already built on top of an RTOS and using those facilities allows integrating Go code (compiled with TinyGo) into an existing project. For example, that way you could replace an existing system written in C/C++ slowly with code written using TinyGo, or use existing libraries that expect to be running on an RTOS. Also, RTOSes often provide their own HAL which could be a quick way to support new hardware.

gordonklaus, 2019-08-04:
Thanks for your answers, @deadprogram and @aykevl.

Migrating existing projects and taking advantage of existing libraries seem like great reasons for integration with Zephyr or other RTOSs.

But I guess what I'm relly interested in is:  What is the planned/desired scope of TinyGo?  Are there features of an RTOS that are not feasible to implement as part the TinyGo ecosystem?  @deadprogram mentioned OTA updates...

Maybe this issue isn't the best place for the discussion.  We could create another issue for including this information in the documentation, if you see fit.  I can imagine a lot the people who are attracted to TinyGo (Go programmers new to embedded – like myself and the OP) would find it valuable.

davidg88007, 2019-11-06:
Zephyr supports OpenThread, enabling mesh networking of devices.  Likely prohibitive to re-implement Thread in TinyGo, whereas using TinyGo on Zephyr devices, would be useful.

dsseng, 2025-08-13:
I'm also interested in this, because Zephyr has quite broad hardware support (not just SoCs but also sensors, protocols, various power management routines). It also looks like newer nRF series like nRF54 do nor have a SoftDevice controller, but only Zephyr-centric ones in Zephyr and NCS, so for those to be used Zephyr is likely highly recommended.

CAN, LVGL, many networking drivers and overall integration into an industry-standard solution could boost TinyGo really well, giving support for hardware for free even when maintainers don't have that hardware.

Given there's an `asyncify` scheduler, using Zephyr scheduling primitives should be possible as well. Zephyr is also capable of supporting Rust as the application language, so there is already some work in the similar direction.

I might be interested in experimenting with this integration opportunity if it is considered