tinygo-org/tinygo #397: Switch to CMSIS 5

QuLogic, 2019-06-07:
The [license](https://github.com/ARM-software/CMSIS/blob/master/CMSIS/CMSIS_END_USER_LICENCE_AGREEMENT.pdf) for CMSIS 4 is rather complicated, mostly due to being a custom license, with separate terms for different parts of the repo. [CMSIS 5](https://github.com/ARM-software/CMSIS_5) is much simpler though, because it's just one license, [Apache 2.0](https://github.com/ARM-software/CMSIS_5/blob/develop/LICENSE.txt).

CMSIS 4 also hasn't been touched in 3.5 years, though I'm unsure if the headers used require all that much care.

It would be nice if tinygo could switch to using CMSIS 5 instead. Unfortunately, I don't have any nrf hardware to test, or I might attempt this myself.

deadprogram, 2019-06-07:
In theory, that sounds nice. But we need to work with vendor provided SVD files for each for the platforms we support, and there are not CMSIS 5 versions of all these files. At least I could not find them anywhere.

aykevl, 2019-06-07:
I don't think the CMSIS version here is related to these SVD files. The main reason why we need CMSIS is for a few header files for the nrf, but not much else (example: we don't use the CMSIS interrupt macros but instead use ones written in Go). That will likely change with improved CGo support, though: existing libraries will likely use CMSIS.

For me, the main reason why I haven't switched the repo over was that CMSIS 5 is _much_ bigger. That's less of a problem now because (I think) more people use the release builds and we probably only need to include a small part of the whole CMSIS system.

QuLogic, 2019-06-07:
You're right, the new one is pretty big to clone. The checkout itself is not too much different:
```
$ du -hsc lib/CMSIS* --exclude=.git
294M	lib/CMSIS
326M	lib/CMSIS_5
```
whereas the repo itself is triple:
```
$ du -hsc .git/modules/lib/CMSIS/ lib/CMSIS_5/.git/ --exclude=lfs
63M	.git/modules/lib/CMSIS/
203M	lib/CMSIS_5/.git/
```
However, they did move a lot of stuff to git-lfs:
```
$ du -hsc lib/CMSIS_5/.git/lfs/
192M	lib/CMSIS_5/.git/lfs/
```
so you can disable it to get a smaller repo:
```
$ du -hsc --exclude=.git CMSIS_5_no_lfs/
94M	CMSIS_5_no_lfs/
$ du -hsc CMSIS_5_no_lfs/.git
204M	CMSIS_5_no_lfs/.git
```
Unfortunately, I don't know if you can set that for a submodule.

QuLogic, 2019-06-12:
Alas, I've still not figured out a way to disable lfs (other than not having it installed or explicitly skipping it during a clone), but if you want to test the clone size, you can try with https://github.com/QuLogic/tinygo by cloning the `cmsis5` branch.

aykevl, 2019-06-13:
@QuLogic slightly unrelated but what are your thoughts on newlib? The license story of that library is really messy but it is used everywhere and I intend to include it in TinyGo at some point as a libc on embedded devices (so you can do `printf` in a C file added with CGo).

QuLogic, 2019-06-14:
It is rather messy, but fortunately, they're all free licenses and this has already been verified as there are packages of it already.

Would this be added on microcontrollers or arm, or both?

aykevl, 2019-09-22:
Whoops forgot to reply here.

> It is rather messy, but fortunately, they're all free licenses and this has already been verified as there are packages of it already.

Glad that this is sorted.

> Would this be added on microcontrollers or arm, or both?

Cortex-M microcontrollers and possibly other microcontrollers: Linux/ARM already has a libc. At the moment, the support for CGo is very weak but if we want to make it usable, there has to be some sort of libc.

QuLogic, 2019-09-26:
I recently came across [picolibc](https://github.com/keith-packard/picolibc), which you might be interested in.

closed 2024-11-27 by deadprogram