tinygo-org/tinygo #410: Should the wasm_exec.js file be updated?

13rac1, 2019-06-18:
https://github.com/tinygo-org/tinygo/blob/da857108/targets/wasm_exec.js is a fork of an older version of the original upstream https://github.com/golang/go/blob/master/misc/wasm/wasm_exec.js file. There are a number of fixes on both sides since the fork. Should it be updated? 

Example of some of the changes:
```diff
--- targets/wasm_exec.js        2019-06-12 15:55:00.307116446 -0700
+++ ../dev-go/misc/wasm/wasm_exec.js    2019-06-17 18:29:30.759958895 -0700
@@ -1,42 +1,36 @@
 // Copyright 2018 The Go Authors. All rights reserved.
 // Use of this source code is governed by a BSD-style
 // license that can be found in the LICENSE file.
-//
-// This file has been modified for use by the TinyGo compiler.
 
 (() => {
-       // Map web browser API and Node.js API to a single common API (preferring web standards over Node.js API).
-       const isNodeJS = typeof process !== "undefined";
-       if (isNodeJS) {
-               global.require = require;
-               global.fs = require("fs");
-
-               const nodeCrypto = require("crypto");
-               global.crypto = {
-                       getRandomValues(b) {
-                               nodeCrypto.randomFillSync(b);
-                       },
-               };
+       // Map multiple JavaScript environments to a single common API,
+       // preferring web standards over Node.js API.
+       //
+       // Environments considered:
+       // - Browsers
+       // - Node.js
+       // - Electron
+       // - Parcel
+
+       if (typeof global !== "undefined") {
+               // global already exists
+       } else if (typeof window !== "undefined") {
+               window.global = window;
+       } else if (typeof self !== "undefined") {
+               self.global = self;
+       } else {
+               throw new Error("cannot export Go (neither global, window nor self is defined)");
+       }

```

deadprogram, 2019-06-22:
Any comments on this from @johanbrandhorst or others working on the WASM side of things?

johanbrandhorst, 2019-06-22:
Sounds like it would be nice to merge back some of the changes in upstream, but a
AFAIK the TinyGo wasm_exec.js has deviated significantly. I am not that familiar with the changes, it'd have to be Ayke who knows if this is possible.

aykevl, 2019-06-24:
Yes, there were quite a few changes to wasm_exec.js to support the different ABI / calling convention of TinyGo. However, some changes would still be nice to copy, like the wider support of environments. Some changes would probably require some adaptation to be usable in TinyGo.

13rac1, 2019-06-24:
I compared this version with multiple historical versions of the original file trying to find the source. I decided I don't have enough context with the project to figure out the correct SHA by looking at them.

deadprogram, 2019-10-14:
Reminder that we still probably need to look at this...

deadprogram, 2020-04-21:
We've done this recently as part of the v0.13.0 release, so now closing. Thanks everyone!

closed 2020-04-21 by deadprogram