tinygo-org/tinygo #411: can not pass bytes array to wasm

fractus-chain, 2019-06-18:
I was run tinygo compiler to compile this code:
```go
//go:export AppMain2
func AppMain2(data []byte) {
	println(string(data[0:]))
}
```
then i run the wasm file by the life webassembly virtualmachine:
```go
	run, ok := vm.GetFunctionExport("AppMain2")
	if !ok {
		panic("function not found: run")
	}

	data := []byte("data from life vm")
	copy(vm.Memory, data)

	if _, err := vm.Run(run, 0, int64(len(data)), int64(cap(data))); err != nil {
		fmt.Println(err)
		os.Exit(1)
	}
```
when the vm was executed, i found an error:
```
panic: runtime error: nil pointer dereference
wasm: unreachable executed
```
but when i get bytes from wasm code, like this:
```go
//go:export AppMain2
func AppMain2(data []byte) {
	println(string(data[1:]))
}
```
everything goes ok, only the first character was missed.

so can you team help me?

aykevl, 2019-06-18:
The problem here is that you are copying the data to address 0. If you copied it to any other address and passed that pointer, it would work out just fine. While address 0 is a legal address in WebAssembly, it is not a legal address in TinyGo (it is a nil pointer and you cannot dereference a pointer that is nil).

However, the bigger problem is that you copy to a random location in linear memory. As far as I know, the linker currently lays out memory in a way that puts global variables first, then the stack, and then the heap. The code above would probably overwrite some global variables or modify the stack. A better way to solve this would be the following code (untested):

```go
var buf [1024]byte

//go:export bufAddr
func bufAddr() *byte {
	return &buf[0]
}

//go:export AppMain2
func AppMain2(data []byte) {
	println(string(data))
}
```

You could then call `bufAddr` first to get a legal pointer, and later call `AppMain2` with the same pointer but the new data.

aykevl, 2019-06-18:
Closing because this is working as intended.

closed 2019-06-18 by aykevl