tinygo-org/tinygo #418: Is there any useable serialize/unserialize tools in tinygo?

fractus-chain, 2019-06-21:
msgpack can not compile in tinygo and there is no builtin tools like encoding/json encoding/gob.

aykevl, 2019-06-21:
Not that I'm aware of, but you can take a look at serialization tools that do not use reflection. Reflection still has poor support in TinyGo (just enough to get `fmt.Println` to work).

diamondburned, 2019-07-03:
Does `encoding/binary` work?

On an unrelated note, how well does `fmt.Println` work?

deadprogram, 2019-07-03:
I make pretty extensive use of `encoding/binary` in the USB CDC implementation for the SAMD21 processors, for example: 
https://github.com/tinygo-org/tinygo/blob/master/src/machine/machine_atsamd21.go

`fmt.Println()` works pretty well, but you will surely find edge cases that need further work. Any use of reflection increases the binary size of the executable so I use `println()` unless I need the formatting, or am already using reflection.

diamondburned, 2019-07-04:
So only `encoding/json` doesn't work? What about https://github.com/francoispqt/gojay, it uses no reflections for decoding, and encoding could be done without any as well.

aykevl, 2019-07-05:
The simplest way to know that is to try it out.

aykevl, 2019-07-05:
(closing this because this is not a bug but a question - feel free to continue the discussion)

closed 2019-07-05 by aykevl