tinygo-org/tinygo #420: Incorrect function signature detection?

justinclift, 2019-06-21:
Just stumbled over this error when testing out the WIP conservative gc:

```
$ tinygo build -target wasm -gc conservative -no-debug -o docs/wasm.wasm wasm.go
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double -2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double 2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double -2.500000e+01, double 0.000000e+00, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 2.500000e+01, double 0.000000e+00, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double -2.500000e+01, double 2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 2.500000e+01, double 2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double -2.500000e+01, double -2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 2.500000e+01, double -2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double 0.000000e+00, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double 0.000000e+00, i8* %task.handle)
error: optimizations caused a verification failure
```

Also happens with the `leaking` gc, so it's probably not a gc thing?

```
$ tinygo build -target wasm -gc leaking -no-debug -o docs/wasm.wasm wasm.go
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double -2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double 2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double -2.500000e+01, double 0.000000e+00, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 2.500000e+01, double 0.000000e+00, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double -2.500000e+01, double 2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 2.500000e+01, double 2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double -2.500000e+01, double -2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 2.500000e+01, double -2.500000e+01, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double 0.000000e+00, i8* %task.handle)
Call parameter type does not match function signature!
  %task.handle = call i8* @llvm.coro.begin(token %task.token, i8* %task.data)
 double  call void @processOperation(i32 0, i32 50, i32 12, double 0.000000e+00, double 0.000000e+00, i8* %task.handle)
error: optimizations caused a verification failure
```

It's interesting the final call param is listed as `*i8.%task.handle`, when it should be a `double` like the two before:

https://github.com/justinclift/tinygo_canvas_test1/blob/2bd7f103d036969e15689cd660f6e0306d6576c8/wasm.go#L286

The code which calls it is this:

https://github.com/justinclift/tinygo_canvas_test1/blob/2bd7f103d036969e15689cd660f6e0306d6576c8/wasm.go#L253-L280

Bug, or am I just too tired again atm? :smile:

aykevl, 2019-06-22:
Yes that's a bug. LLVM errors are always a compiler bug. The compiler should either emit the correct code or show a proper error message.

Can you try reducing this testcase to something minimal that still triggers this error? That would save me time fixing this bug :)

justinclift, 2019-06-22:
Yep, will do. :smile:

justinclift, 2019-06-22:
As you mentioned on Slack, it was a case of mixing a blocking operation with an exported function.

The exported function was calling `time.Sleep()`, which I hadn't noticed. :frowning:

closed 2019-06-22 by justinclift