tinygo-org/tinygo #421: Possibly spurious error: LLVM ERROR: Coroutines cannot handle non static allocas yet

justinclift, 2019-06-23:
Came across an interesting LLVM ERROR while adjusting the repulsion demo to use JS timers instead of Go created callbacks.

```
$ tinygo build -target wasm -gc conservative -no-debug -o html/test.wasm canvas/main.go
LLVM ERROR: Coroutines cannot handle non static allocas yet
```

Instead of this being a cut-n-dried case of something being wrong with the code, this is turning up after **working** code (that compiles fine) has a non-useful part commented out.  eg there's no functional change to the code.  No change in what touches globals, etc.

Seems like something isn't quite right yet. :wink:

The full code (compiles ok):

The code section to comment out, to trigger this bug:

justinclift, 2019-06-23:
Note - the easiest way to workaround the problem is to remove the `chan` completely.  Doesn't seem to be needed these days anyway. :wink:

aykevl, 2019-06-24:
That's interesting. It is probably still caused by the `defer` somehow (the second?) because defer does a stack allocation. The easy way to fix this is to use heap allocation, but most defer statements can easily be stack allocated.

A similar problem existed in the gc toolchain but seems to be fixed now: https://github.com/golang/go/issues/14939

justinclift, 2019-06-24:
> The easy way to fix this is to use heap allocation ...

Are you meaning create `mouseMoveEvt` using `new()`?

deadprogram, 2019-09-17:
Is this issue and #234 the same issue? Also, is this still an issue with the latest `dev` branch?

justinclift, 2019-09-17:
Not sure atm.  I'll have time to get back into TinyGo at the start of next month, so hopefully it can wait until then? :smile:

jadr2ddude, 2019-09-27:
This remains a major limitation. For those who do not know what a static alloca is, a static alloca is a stack allocation that occurs in the entry basic block. Therefore, if the alloca is outside the entry block then we cannot do coroutine lowering at the moment. While implementing blocking select, I ran into this. This definitely is present right now.

aykevl, 2019-09-30:
Allocas should really be moved into the entry block if possible (which is almost always the case). Take a look here for how it can be done: https://github.com/tinygo-org/tinygo/pull/359

jaddr2line, 2020-03-01:
After #828, there should not be any generated non-static allocas left.

aykevl, 2020-03-01:
Closing this issue too, see https://github.com/tinygo-org/tinygo/issues/234#issuecomment-593136349.

closed 2020-03-01 by aykevl