tinygo-org/tinygo #422: Status of support for maps

lei-april, 2019-06-24:
According to https://tinygo.org/lang-support/#maps:
> A map can store at most 8 entries. The exception is when it is initialized as a global variable, in which case it can store more entries.

What would exactly happen if I initialize a map as a local variable, and then insert more than 8 entries into it?

I experimented with the following code and it seems to work:
```go
package main

import "strconv"

func main() {
    m := make(map[string]string)
    for i := 0; i < 100; i++ {
        k := "key" + strconv.Itoa(i)
        v := "val" + strconv.Itoa(i)
        m[k] = v
    }

    for _, v := range m {
        println(v)
    }
}
```

aykevl, 2019-06-24:
Ah yes, the docs are outdated. Maps are now better supported and should now work for the most common cases. However, they're still pretty slow for larger maps: lookups happen linearly unless you pass the right size when you make the map (for example, with `make(map[string]string, 100)`).

Leaving this issue open as a reminder that the docs need to be updated.

deadprogram, 2019-07-08:
This commit updated the docs https://github.com/tinygo-org/tinygo-site/commit/b80ca669e0b7f636e48a350210a3f1ece9dcc863 I an going to close this issue now. Please reopen if needed. Thank you!

closed 2019-07-08 by deadprogram