tinygo-org/tinygo #424: bytealg.CountString is not supported

1l0, 2019-06-27:
main.go:
```go
package main

import (
	"fmt"
	"strings"
)

func main() {
	s := strings.Replace("abc", "a", "x", -1)
	fmt.Println(s)
}
```
run:
```sh
$ tinygo run main.go
Undefined symbols for architecture x86_64:
  "_internal/bytealg.CountString", referenced from:
      _main in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
error: failed to link /var/folders/kt/gfbj96zj1jvd9tsbbfjcg94h0000gn/T/tinygo175846444/main: exit status 1
$ go run main.go
xbc
```
How can I fix this?

aykevl, 2019-06-27:
A related function is implemented here in the runtime, which you can use as an example:

https://github.com/tinygo-org/tinygo/blob/40b193f1faa507e8d6407624c1f00937602b0b89/src/runtime/string.go#L184-L194

You could either implement your own version, or copy the one from the Go standard library:
https://github.com/golang/go/blob/master/src/internal/bytealg/count_generic.go
But note that if you copy from the standard library, you have to clearly state where it came from.

1l0, 2019-06-27:
I wonder if I have understood correctly.
Should following be compiled by TinyGo (actually failed) or do I have to compile TinyGo itself with a modification?
```go
package main

import (
	"fmt"
	"strings"
	_ "unsafe"
)

func main() {
	s := strings.Replace("abc", "a", "x", -1)
	fmt.Println(s)
}

// paste from https://github.com/golang/go/blob/67f181bfd84dfd5942fe9a29d8a20c9ce5eb2fea/src/internal/bytealg/count_generic.go#L19
//go:linkname countString internal/bytealg.CountString
func countString(s string, c byte) int {
	n := 0
	for i := 0; i < len(s); i++ {
		if s[i] == c {
			n++
		}
	}
	return n
}
```

aykevl, 2019-07-14:
Sorry for the late reply.

Outside of the runtime, `//go:linkname` doesn't always work. This is a bug I'm aware of and should be fixed in the not too distant future hopefully (see #285). So simply moving this function to the runtime should get it to compile.

1l0, 2019-07-15:
OK I will wait the party.

deadprogram, 2019-09-18:
I think what @aykevl was saying @1l0 was that if you add that code to the correct place in the `runtime` will let it compile. You can then submit a PR and this should be solved.

The `//go:linkname` directive does not work on code outside of the `runtime` but works fine when used *within* the `runtime`.

Thanks!

aykevl, 2019-09-18:
Exactly, what @deadprogram says. There is no reason to wait, just move `countString` into the runtime to get it to compile (where it belongs anyway).

1l0, 2019-09-19:
Do you prefer that route (I mean patchwork-like)? I'm not hurrying so I can wait until fixing #285 that seems to fundamentally solve the problem.

deadprogram, 2019-09-19:
These seems some confusion. 

The bug mentioned has nothing to do with the `bytealg.CountString` feature. it is that you cannot use the `//go:linkname` directive outside of the `runtime` package, which is not really related to the `bytealg.CountString` feature in this issue.

#285 has nothing directly to do with your issue.

I hope that clarifies.

deadprogram, 2019-10-10:
My own PR that implemented this based on your research @1l0 was merged into `dev` thanks for working on this. Now closing.

closed 2019-10-10 by deadprogram