tinygo-org/tinygo #425: unable to build due to sys/unix (but sys/unix not used)

tobowers, 2019-06-27:
I'm sorry I don't have an easy reproduction for this (it's a private repo and a complicated build path). However I can show it happens below.

The meat of it is that it believes one of my repos has a dependency on x/sys/unix but it doesn't (as proven by the `go mod graph | grep unix` below). 

The only thing I can think of is that the *master* branch of that repo *does* have a dependency on x/sys/unix . I tried to vendor everything as well, but produced the same errors.

```
tinygo version
tinygo version 0.6.0 darwin/amd64
```

```
14:16 ~/code/tupelo-go-sdk/wasm (play/wasm) $ tinygo build -o wasm.wasm -target wasm ./main.go
error: no buildable Go source files in /Users/tobowers/code/go/src/github.com/quorumcontrol/chaintree/vendor/golang.org/x/sys/unix
14:16 ~/code/tupelo-go-sdk/wasm (play/wasm) $ tinygo clean
14:16 ~/code/tupelo-go-sdk/wasm (play/wasm) $ tinygo build -o wasm.wasm -target wasm ./main.go
error: no buildable Go source files in /Users/tobowers/code/go/src/github.com/quorumcontrol/chaintree/vendor/golang.org/x/sys/unix
14:16 ~/code/tupelo-go-sdk/wasm (play/wasm) $ go mod graph | grep unix
14:16 ~/code/tupelo-go-sdk/wasm (play/wasm) $  GOOS=js GOARCH=wasm go run .
we did all the things
```

tobowers, 2019-06-27:
The last line there is showing (non-tiny) go is able to compile and execute the wasm and get the output ('we did all the things')

aykevl, 2019-06-27:
There probably still is a dependency _somewhere_. TinyGo won't add a random dependency for no reason.

Note that we do not support Go modules yet, everything is GOPATH based (and not even $GOPATH/pkg/mod). This may confuse things.

Better diagnostics are definitely a priority (goal: better diagnostics than the `go` tool) but we're not there yet, unfortunately.

tobowers, 2019-06-27:
Oh... well just looking at GOPATH is probably the problem, right? Because the master branch *does* have the sys/unix dependency. How should I version the dependencies for tinygo?

aykevl, 2019-06-27:
Versioning dependencies in TinyGo is the same as versioning dependencies in Go before `go mod` (that is, basically unsupported). I'm not sure whether TinyGo supports a vendor directory but if it does, you can try dep.

Of course, we will want to support versioning eventually as it is very important. But it requires some major refactoring that is not yet completed, see #285.

tobowers, 2019-06-27:
Thank you. Yeah I tried a vendor and it didn’t work.
-- 
Topper Bowers

topper@toppingdesign.com
github.com/tobowers
linkedin.com/in/topperbowers

deadprogram, 2024-11-18:
Closing due to age. Thanks to everyone!

closed 2024-11-18 by deadprogram