tinygo-org/tinygo #431: compilation error?

grd, 2019-07-01:
Hi! When I compile with `$ tinygo flash -target=arduino examples/blinky1` I get the following answer:

error: cannot find package "errors" in any of:
	/home/user/go/src/github.com/tinygo-org/tinygo/src/errors (from $GOROOT)
	/snap/go/3129/src/errors (from $GOPATH)
	/home/user/go/src/errors

To solve the problem I edited .profile like `export GOPATH=/snap/go/current` and then I got a whole bunch of errors.

I am on xubuntu x64 but I doubt that that is the source of it. This silly problem has become rather big to me. And I have to admit that the installation procedure has improved massively ;-)

aykevl, 2019-07-01:
In general, it is best to not set `GOROOT`. In most cases, it gets configured automatically.

Can you share:

* What installation method you used for TinyGo (manual build, install from .deb, etc.)
* `which go`
* `go env`

That will help diagnose the issue.

grd, 2019-07-01:
The installation was with snap. Snap version is 3947. Which go is dependent on snap and is /snap/bin/go. The version of Go (with `go version`) is `go version go1.12.6 linux/amd64`. The output of `go env` is `
GOARCH="amd64"
GOBIN=""
GOCACHE="/home/user/.cache/go-build"
GOEXE=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOOS="linux"
GOPATH="/home/user/go"
GOPROXY=""
GORACE=""
GOROOT="/snap/go/3947"
GOTMPDIR=""
GOTOOLDIR="/snap/go/3947/pkg/tool/linux_amd64"
GCCGO="gccgo"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD=""
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build931523382=/tmp/go-build -gno-record-gcc-switches"

aykevl, 2019-07-01:
There are multiple ways to install according to the docs, did you install using the first - that is, using the deb package?

aykevl, 2019-07-01:
> `tinygo version 0.2.0 linux/amd64`

That's a really old version! Can you update to the last version (v0.6.0)? There have been lots of improvements that are likely to fix the problem you have.

grd, 2019-07-01:
Thanks! Now I have got a new problem. I look like a n00b around here... 
I installed it with a .deb file. `tinygo version` output is `tinygo version 0.6.0 linux/amd64`. So I guess that it's okay. Now I put in an arduino and I call it with the following command `$ tinygo flash -target=arduino examples/blinky1` And the output is:
`avrdude: ser_open(): can't open device "/dev/ttyACM0": Permission denied

avrdude done.  Thank you.

error: failed to flash /tmp/tinygo995484237/main.hex: exit status 1
user@idea:~$ tinygo version
tinygo version 0.6.0 linux/amd64`

Thanks a lot!

aykevl, 2019-07-01:
That's not a problem of TinyGo but of the configuration of your system. I don't know exactly what it is but you should be able to figure this out with your favorite search engine. For example, take a look here: https://arduino.stackexchange.com/questions/21215/first-time-set-up-permission-denied-to-usb-port-ubuntu-14-04

grd, 2019-07-02:
Thanks a lot! Now I think that this can be closed.

closed 2019-07-02 by grd