tinygo-org/tinygo #433: Docker: could not read version from GOROOT: /usr/local/go

diamondburned, 2019-07-02:
Hello. Was grabbing `tinygo` manually when I had this:

```sh
$ tinygo build -o ../sc.wasm -target wasm32-unknown-unknown .
error: could not read version from GOROOT: /nix/store/ypfhc6f07z6ij7yn5hrz87s74jh7l8w2-go-1.12.1/share/go
```

Tried Docker as well:

```sh
$ docker run --rm -v "$PWD" tinygo/tinygo tinygo build -o ../sc.wasm -target wasm32-unknown-unknown .
Unable to find image 'tinygo/tinygo:latest' locally
latest: Pulling from tinygo/tinygo
11f34d1ed4d8: Pull complete 
b74da5bb8068: Pull complete 
babdd2d5058f: Pull complete 
f913f1fc17b5: Pull complete 
698bf23304a5: Pull complete 
4e4c4039cb7f: Pull complete 
fe88c201175d: Pull complete 
a27575ff07ae: Pull complete 
Digest: sha256:07dcaa92012e5f0b8c4e138db799e54a9f1b1729f8aa16c85d4232a9017dcea4
Status: Downloaded newer image for tinygo/tinygo:latest
error: could not read version from GOROOT: /usr/local/go
```

In case it matters, the repository is [diamondburned/smart-contract-go](https://github.com/diamondburned/smart-contract-go).

iwasaki-kenta, 2019-07-02:
Got the same issue as well on tinygo 0.6.0. It looks like it is looking for a `VERSION` file which does not exist with quite a number of Go distributions.

[image]

deadprogram, 2019-07-03:
@aykevl any thoughts on this?

lei-april, 2019-07-03:
I had the same issue on macOS. TinyGo works with the standard Go installed by Homebrew, but not with the one installed by MacPorts, due to the missing `VERSION` file.

aykevl, 2019-07-03:
Maybe we should then parse src/runtime/internal/sys/zversion.go instead like GoLand seems to do (CC @justinclift).
Patches welcome :smile: you can use the [go/parser](https://golang.org/pkg/go/parser/#ParseFile) package and look for the `TheVersion` declaration.

---

I would argue that it's a problem with the distribution but it's much easier to fix it ourselves than fix the distributions.

justinclift, 2019-07-03:
Do we used the parsed version for anything important?  Kind of thinking we could just detect the present or appropriate dirs (eg `src`, `targets`), and call it "good enough".  The version # check could be a warning instead of a fatal error?

aykevl, 2019-07-03:
Yes, the version number is important. We use it to set `go1.1`, `go1.2`, ... `go.1.12` build tags.

diamondburned, 2019-07-04:
What's wrong with `exec.Command("go", "version")`?

diamondburned, 2019-07-04:
Holding onto the PR, I didn't notice the dev branch.

diamondburned, 2019-07-04:
Fixed it to obey the contribution guidelines.

aykevl, 2019-07-05:
> What's wrong with `exec.Command("go", "version")`?

The problem is that the function tries to read the Go version of a given GOROOT, and this command returns the version of whatever `go` tool there is in $PATH, not necessarily of the passed-in GOROOT (you can have multiple go installations). Also, running an external program is slower than necessary.

diamondburned, 2019-07-05:
Removed `go version` check  (though I personally still don't see an issue, I just removed it for the sake of this PR)

aykevl, 2019-07-06:
Fixed by #435.

tobowers, 2019-07-10:
this seems like it's still broken in the docker image, any chance we can get a release?

aykevl, 2019-07-10:
We will have one next week.

gertcuykens, 2020-01-18:
Using `tinygo` dev branch and doesn't work when you use the `go` master branch in `/usr/local/go`
```sh
% tinygo build -o a.out
error: could not read version from GOROOT (/usr/local/go): could not parse Go version: version does not start with 'go' prefix
```

```sh
% go version             
go version devel +c170b14c2c Tue Dec 24 16:13:37 2019 +0000 darwin/amd64
```

tegk, 2020-08-11:
why was this closed? It still does not work with Ubuntu and go installed via snap

quaintdev, 2021-01-09:
I am also facing the issue with Ubuntu and snap. Any suggestions?

justinclift, 2021-01-09:
As a general thought, doesn't Snap sandbox stuff by default?  eg only allow access to the user's home directory, unless specifically configured otherwise in the application's Snap config?

aykevl, 2021-03-11:
The issue with Go installed from snap must be a different issue. Closing this, see #1706 for the Snap incompatibility.

closed 2021-03-11 by aykevl

helloworlde, 2021-09-03:
Same issue on MacOS installed by brew:
```
tinygo version
tinygo version 0.19.0 darwin/amd64 (using go version <unknown> and LLVM version 11.0.0)

go version
go version go1.17 darwin/amd64

which go
/usr/local/opt/go/libexec/bin/go
```
and GOROOT is `GOROOT="/usr/local/opt/go/libexec"`

indiependente, 2021-09-03:
Same issue for me as well on macOS
```
➜  ~ go version
go version go1.17 darwin/amd64
➜  ~ tinygo version
tinygo version 0.19.0 darwin/amd64 (using go version <unknown> and LLVM version 11.0.0)
➜  ~ which go
/usr/local/bin/go
➜  ~ go env | grep GOROOT
GOROOT="/usr/local/Cellar/go/1.17/libexec"
```

deadprogram, 2021-09-03:
See https://github.com/tinygo-org/tinygo/issues/2080

Fl4v, 2021-10-02:
Not sure if it helps, but I resolved this issue by installing the latest version of `tinygo` on Ubuntu:

```bash
wget https://github.com/tinygo-org/tinygo/releases/download/v0.20.0/tinygo_0.20.0_amd64.deb
sudo dpkg -i tinygo_0.20.0_amd64.deb
```

aykevl, 2021-10-02:
@Fl4v yes, that's issue #2080 that you were hitting.

If anyone still encounters this issue in TinyGo 0.20.0 or later, please open a new issue instead of reopening this one.

Fl4v, 2021-10-02:
Thank you @aykevl

commit aabb6ba2 (2019-07-06): main: use zversion.go and VERSION files to detect version, fixes #433
commit 774c86e2 (2021-03-14): goenv: use physical path instead of cached GOROOT in function getGoroot (fixes #433, #1658)