tinygo-org/tinygo #436: runtime: Garbage collector should not recurse

johanbrandhorst, 2019-07-04:
At least within the context of WebAssembly, the conservative GC recursing is causing stack overflows on larger heaps. Ideally it should be possible to make it not recurse.

aykevl, 2019-07-04:
This will also be an issue on embedded devices but hasn't been so far.

---

The MicroPython GC has avoided recursion by using a fixed-size pointer stack and scanning the complete heap in the worst case, which avoids stack overflow but is not very efficient. I'm not aware of a more efficient algorithm and I doubt one can exist in a simple mark-sweep collector.

CryZe, 2019-07-04:
iirc tinygo only uses a single wasm page (of 64 KiB) for the stack. It might make sense to bump that value in addition / as an alternative to rewriting the GC.

jaddr2line, 2020-04-10:
The fix #1028 has been merged onto `dev` and will land in v0.13.

PR #1028 (2020-04-09): runtime (gc): remove recursion from "conservative" GC
This PR attempts to fix #436 and hopefully reduce stack usage.

The GC modifications take a similar approach to the MicroPython GC, in that a fixed-size stack is used to track allocations that need to be scanned. If the stack overflows, this forces a full rescan of all marked allocations. This is potentially a lot slower in some extreme cases, but does not blow up the stack like the current version does.

This seems to pass some basic sanity checks, but we should try this out with some larger tinygo programs before merging it.
closed 2020-04-10 by jaddr2line