tinygo-org/tinygo #437: error: interp: branch on a non-const-propagated constant expression

diamondburned, 2019-07-05:
This error happened when compiling https://github.com/diamondburned/smart-contract-go/blob/master/examples/transfer-back/main.go with `/tinygo build -o ./main.wasm`.

With `-printir`: http://0x0.st/zLo5.txt

`error: interp: branch on a non-constant` is returned with `-gc none`.

Since Google suggests something about `const`s not being supported, I wrote some smaller code to confirm that, but couldn't.

diamondburned, 2019-07-05:
Output of `llc -version`:

```sh
LLVM (http://llvm.org/):
  LLVM version 8.0.0
  Optimized build.
  Default target: x86_64-unknown-linux-gnu
  Host CPU: skylake

  Registered Targets:
    aarch64    - AArch64 (little endian)
    aarch64_be - AArch64 (big endian)
    amdgcn     - AMD GCN GPUs
    arm        - ARM
    arm64      - ARM64 (little endian)
    armeb      - ARM (big endian)
    bpf        - BPF (host endian)
    bpfeb      - BPF (big endian)
    bpfel      - BPF (little endian)
    hexagon    - Hexagon
    lanai      - Lanai
    mips       - MIPS (32-bit big endian)
    mips64     - MIPS (64-bit big endian)
    mips64el   - MIPS (64-bit little endian)
    mipsel     - MIPS (32-bit little endian)
    msp430     - MSP430 [experimental]
    nvptx      - NVIDIA PTX 32-bit
    nvptx64    - NVIDIA PTX 64-bit
    ppc32      - PowerPC 32
    ppc64      - PowerPC 64
    ppc64le    - PowerPC 64 LE
    r600       - AMD GPUs HD2XXX-HD6XXX
    sparc      - Sparc
    sparcel    - Sparc LE
    sparcv9    - Sparc V9
    systemz    - SystemZ
    thumb      - Thumb
    thumbeb    - Thumb (big endian)
    wasm32     - WebAssembly 32-bit
    wasm64     - WebAssembly 64-bit
    x86        - 32-bit X86: Pentium-Pro and above
    x86-64     - 64-bit X86: EM64T and AMD64
    xcore      - XCore
```

Output of `tinygo version`:

```
tinygo version 0.7.0-dev linux/amd64
```

If it helps:

```nix
{ pkgs ? import <nixpkgs> {} }:

let
  unstable = import <unstable> {};
in pkgs.stdenv.mkDerivation rec {
	name = "perlin";

	buildInputs = with unstable; [
		cquery emscripten wabt llvm-polly lld_8
		llvmPackages.clang-unwrapped clang-polly
	];
}
```

aykevl, 2019-07-06:
Thank you for reporting! This is a bug in the interp package and I know roughly what to look for.

aykevl, 2019-07-06:
Unfortunately, I can't access the gitlab.com/diamondburned/smart-contract-go package. Assuming it can't be made public, can you share a reduced test case? The issue lies in the initialization code of the package or one of its dependencies, so you can test in which package it lies simply by importing it.

Also, is there a reason why you pass `-gc=dumb`?

diamondburned, 2019-07-06:
I made a mistake, it was `github`. I just passed in `-gc=dumb` to test it.

diamondburned, 2019-07-18:
Are there any updates on this?

deadprogram, 2019-09-17:
Hi @diamondburned you might want to try the current `dev` branch of TinyGo as it probably addresses this issue.

deadprogram, 2019-09-20:
As of TinyGo v0.8.0 several packages including `encoding/base64` still cannot be imported due to this error, so this is still not fixed.

deadprogram, 2019-09-20:
https://tinygo.org/lang-support/stdlib/#encoding-base64 for details

bradleypeabody, 2019-11-16:
FYI importing `crypto/rand` also causes `error: interp: branch on a non-constant` on the latest tinygo-dev.

aykevl, 2019-11-17:
Looks like this is the line to blame:

https://github.com/golang/go/blob/6ba0be16391697bdaab7d8f836b66921977d5474/src/math/big/float.go#L559

aykevl, 2019-11-17:
Apparently it's a problem with `runtime.isnil`, a trick to work around an important missed optimization in LLVM 8 (in short, nil pointer checking would almost entirely break escape analysis). This is now unnecessary: a fix for this has been merged in LLVM 9. So I hope that once we change to the right way of doing it (plain `icmp` and special function attributes that are valid in Go), this bug will be fixed at the same time.

deadprogram, 2019-11-26:
This should now be corrected in the `dev` branch, since we are using LLVM 9.

aykevl, 2019-11-26:
No. LLVM 9 has a new optimization that allows us to avoid the `runtime.isnil` workaround. But implementing this takes more than just switching to LLVM 9, unfortunately.

asdine, 2020-01-13:
Any plan regarding this issue?

firelizzard18, 2020-02-20:
I'm trying to translate [this function](https://github.com/PaulStoffregen/cores/blob/master/teensy3/mk20dx128.c#L1263) from C/GCC to TinyGo, as part of #863. I get this error when I use `arm.AsmFull` and the `mrs` instruction:

```golang
AsmFull("mrs {dest}, faultmask", map[string]interface{}{"dest": &faultmask})
if faultmask != 0 {
	return -1
}
```

aykevl, 2020-10-18:
See https://github.com/tinygo-org/tinygo/pull/1430 for a PR that tries to solve this (but isn't ready yet).

aykevl, 2021-01-09:
#1430 has been merged so this will be fixed in the next release.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

PR #581 (2019-09-24): interp: add tests and implement runtime.sliceCopy
Two changes that depend on each other. I can split out the PR if that makes it easier to review.

* Add tests. Start out with a basic test to see whether interp roughly works.
* implement `runtime.sliceCopy` in the interp package. That function uses lots of pointer casting so it's hard to interpret it directly, therefore I implemented it directly in interp.

This change gets 3 packages to compile: encoding/base32, encoding/base64, and encoding/pem. With that, compiling encoding/json is one step closer.

Fixes #437.
closed 2021-03-08 by deadprogram