tinygo-org/tinygo #441: installation instructions

pannous, 2019-07-07:
so I tried to install tiny go ...

https://tinygo.org/
Ready to get started? Click on.

not clickable lol

ok found this:
https://github.com/tinygo-org/tinygo

no direct installation instructions but I found this link back:
https://tinygo.org/getting-started/

To use TinyGo, you must install it on your system. We have instructions for Linux, macOS, Windows, and Docker. You can also install TinyGo from source.

ok by now I am sure that you're making fun of me.

oh wait there is one more tiny link in the navigation bar
https://tinygo.org/getting-started/linux/

deadprogram, 2019-07-07:
Hello @pannous 

Apparently "Click on." was too much a figure of speech., sorry about the confusion. It meant to continue clicking, not to click on a specific link.

So what it would seem you are referring to is also discussed in https://github.com/tinygo-org/tinygo-site/issues/27 which is probably better place to have created this issue. Although it is probably a duplicate.

Anyhow, thanks for reporting, we will merge the other PR soon.

deadprogram, 2019-08-07:
This was merged a few weeks ago, sorry I forgot to note that. Now closing, thanks again.

closed 2019-08-07 by deadprogram