tinygo-org/tinygo #447: Tinygo need serialization tools

fractus-chain, 2019-07-08:
We really need serialization tools for tinygo, because tiny go can not use serialization tools just like coding/json encoding/gob in official golang.

m-chichikalov, 2019-07-08:
Agree, having Google ProtoBuf is must have from the embeded world point of view... I didn't try to compile ProtoBuf with TinyGo so far.

PaluMacil, 2019-09-19:
From the list of things that work and what they are blocked on, it looks like to enable a lot of them we need to get base64 working first before moving on to solve gob, etc. The error for base64 regarding branching on a const expression might be addressed in dev, according to a comment on #437 two days ago, so this is a good place to look. If this fixed base64, it probably unblocks work on other packages. Gob and json might benefit from the recent updates to reflection, so it's certainly something worth checking on dev so that we can get going with more encoders.

deadprogram, 2019-09-20:
As of TinyGo v0.8.0 the `encoding/base64` package still cannot be imported due to #437 so that is still not fixed.

PaluMacil, 2019-10-30:
The 0.9 release gave us base32, base64, and csv. We'll probably need additional reflection support before json, xml, and gob. I expect ProtoBuf will have other things it needs. Great progress!

Gregory-Ledray, 2020-02-28:
It's not necessary to use encoding/json to parse and encode JSON in Go. This library compiles on tinygo with -target wasm: https://github.com/buger/jsonparser

This is possible because it import graph is really small - the only thing that imports 'reflect' is its fmt dependency: https://godoc.org/github.com/buger/jsonparser?import-graph

I wasn't able to test jsonparser because I'd like to use it to handle a HTTP request response, but alas I don't see any examples for passing values from JS to exported WASM functions, even though I can call them in the JS callback.

PaluMacil, 2020-02-28:
This is a good tip. While most people using JSON probably don't want the tradeoffs in jsonparser, I'd be willing to bet most users of tinygo _would_ want those tradeoffs, at least if they are using tinygo for tiny machines.

benma, 2020-03-12:
for golang/protobuf, apart from the json issue mention above, there are the following other issues:

```
../../vendor/github.com/golang/protobuf/proto/extensions.go:62:46: Locker not declared by package sync
../../vendor/github.com/golang/protobuf/proto/extensions.go:82:71: Locker not declared by package sync
../../vendor/github.com/golang/protobuf/proto/extensions.go:155:78: Locker not declared by package sync
../../vendor/github.com/golang/protobuf/proto/text_parser.go:765:20: Append not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/text_parser.go:785:18: Append not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/text_parser.go:469:14: cannot convert nil (untyped nil value) to reflect.Type
../../vendor/github.com/golang/protobuf/proto/text_parser.go:506:8: sv.FieldByName undefined (type reflect.Value has no field or method FieldByName)
../../vendor/github.com/golang/protobuf/proto/text_parser.go:507:8: sv.FieldByName undefined (type reflect.Value has no field or method FieldByName)
../../vendor/github.com/golang/protobuf/proto/text_parser.go:561:18: Append not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/text_parser.go:599:21: MakeMap not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/text_parser.go:601:34: dst.Type().Key undefined (type reflect.Type has no field or method Key)
../../vendor/github.com/golang/protobuf/proto/text_parser.go:655:8: dst.SetMapIndex undefined (type reflect.Value has no field or method SetMapIndex)
../../vendor/github.com/golang/protobuf/proto/text.go:404:23: st.Field(i).Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/text.go:412:38: inner.Type().Field(0).Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/text.go:208:13: sv.FieldByName undefined (type reflect.Value has no field or method FieldByName)
../../vendor/github.com/golang/protobuf/proto/text.go:209:12: sv.FieldByName undefined (type reflect.Value has no field or method FieldByName)
../../vendor/github.com/golang/protobuf/proto/text.go:221:11: cannot convert nil (untyped nil value) to reflect.Type
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:1736:10: t.Key undefined (type reflect.Type has no field or method Key)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:1738:44: f.Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:1739:44: f.Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:1799:18: MakeMap not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:1803:5: m.SetMapIndex undefined (type reflect.Value has no field or method SetMapIndex)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:488:57: t.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:610:47: t.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:613:46: t.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:454:39: f.Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:294:60: f.Type.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:302:66: f.Type.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:305:13: f.Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:313:58: f.Type.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:322:18: f.Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:329:17: f.Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:332:51: t.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:366:35: PtrTo not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:378:31: f.Tag.Get undefined (type string has no field or method Get)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:406:29: PtrTo not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:406:39: reflect.Zero(reflect.PtrTo(t)).MethodByName undefined (type reflect.Value has no field or method MethodByName)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:409:71: t.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_unmarshal.go:190:41: PtrTo not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/table_merge.go:212:59: tf.Name undefined (type reflect.Type has no field or method Name)
../../vendor/github.com/golang/protobuf/proto/table_merge.go:500:53: tf.Name undefined (type reflect.Type has no field or method Name)
[... cut ...]
../../vendor/github.com/golang/protobuf/proto/clone.go:222:20: AppendSlice not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/clone.go:227:21: Append not declared by package reflect
../../vendor/github.com/golang/protobuf/proto/clone.go:122:11: in.FieldByName undefined (type reflect.Value has no field or method FieldByName)
../../vendor/github.com/golang/protobuf/proto/clone.go:128:7: out.FieldByName undefined (type reflect.Value has no field or method FieldByName)
```

elliots, 2020-04-22:
For anyone else trying to get protobuf to work... it can. Sort of. 

If you use gogoproto, you can have it generate marshal and unmarshal functions that work on the bytes, without reflection.

```
protoc --gogofaster_out=. *.proto
```

Then, you "just" need to stub the proto package with something like this

```go
package proto

import "errors"

const Marshal = 0
const GoGoProtoPackageIsVersion3 = 0

type Message interface {
}

type InternalMessageInfo interface {
	DiscardUnknown(interface{})
	Merge(interface{}, interface{})
	Marshal(interface{}, interface{}, interface{}) ([]byte, error)
}

func RegisterFile(string, []byte) {}

func RegisterType(interface{}, string) {}

func CompactTextString(interface{}) string {
	return "nope"
}
```

You can then call Marshal and Unmarshal directly.

jsimnz, 2021-03-28:
Just wanted to give a heads up. My tests have found that https://github.com/valyala/fastjson compiles successfully and runs with the TinyGo compiler targeting WASM/WASI.

dosgo, 2021-09-24:
Is there any other way to compile those packages that are not supported by tinygo? Must find support instead?

jsimnz, 2021-09-29:
> Is there any other way to compile those packages that are not supported by tinygo? Must find support instead?

The issue is the TinyGo compiler doesn't support 100% of Go's `reflect` package. And most JSON packages depend on those functions in some way. 

So at the moment, the only work around is find JSON packages that don't need those unimplemented functions.

dosgo, 2021-09-29:
> > 有没有其他方法可以编译tinygo不支持的那些包？必须寻求支持吗？
> 
> 问题是 TinyGo 编译器不支持 100% 的 Go`reflect`包。大多数 JSON 包在某种程度上依赖于这些函数。
> 
> 因此，目前唯一的解决方法是找到不需要那些未实现功能的 JSON 包。

I am a third-party package using the xml package, which is not easy to replace

dosgo, 2021-09-29:
I am a third-party package using the xml package, which is not easy to replace.

jsimnz, 2021-09-29:
Unfortunately, there isn't much that can be done right now until the necessary changes/fixes are made within the TinyGo compiler.

dgryski, 2021-12-01:
The `tinylib/msgp` messagepack code generation package works with tinygo.  There's even a test in the msgp repo for it: https://github.com/tinylib/msgp/tree/master/tinygotest

taowen, 2021-12-04:
I made a proof of concept demo to use code generation to implement `json.Unmarshal` https://github.com/json-iterator/tinygo Is there similar tools already? Is this path worth pursuing?

dgryski, 2021-12-07:
@taowen Yes please!

taowen, 2021-12-09:
@dgryski  all features of `json.Unmarshal` has been implemented https://github.com/json-iterator/tinygo 
verified it is working properly in tinygo https://github.com/json-iterator/tinygo/tree/main/demo

dgryski, 2021-12-17:
It looks like `mailru/easyjson` works with tinygo if you remove the `var  _ easyjson.Marshaler` line from the generate code that wants to bring in `net/http` and friends.

kyleconroy, 2022-02-01:
Since https://github.com/tinygo-org/tinygo/pull/2314 was merged, `mailru/easyjson` now compiles and works.

ppedziwiatr, 2022-03-07:
> Since #2314 was merged, `mailru/easyjson` now compiles and works.

Is there a chance to make it work on macos / M1? 
I'm getting https://github.com/tinygo-org/tinygo/pull/2314#issuecomment-979148013
Removing `var _ easyjson.Marshaler` https://github.com/tinygo-org/tinygo/issues/447#issuecomment-997022972,
but it's not super convenient ;-)

aykevl, 2022-03-07:
> Is there a chance to make it work on macos / M1?
> I'm getting [#2314 (comment)](https://github.com/tinygo-org/tinygo/pull/2314#issuecomment-979148013)

The error you've linked is for x86_64, not arm64.
If you're only getting that error, it's unrelated to serialization and seems to be caused by cryptograpy-related packages (which might be imported by `mailru/easyjson`, I didn't check). For those errors, see: https://github.com/tinygo-org/tinygo/pull/2688

jcchavezs, 2022-03-29:
I haven't tried `encoding/json` yet but from the website I see:

[image]

Which is correct?

Ping @mathetake @jptosso

nickwesselman, 2022-03-31:
Attempting to use `encoding/json` resulted in the following error for me:

```
panic: unimplemented: (reflect.Type).NumMethod()
```

paralin, 2022-06-03:
https://github.com/planetscale/vtprotobuf ?

cbrake, 2022-07-11:
some experiments: https://github.com/cbrake/tinygo

karmem at least compiles. Could not get any protobuf stuff to compile -- may need to port nanopb or something ...

paralin, 2022-09-14:
@cbrake VTProtobuf with the references to the reflection-based Google protobuf libraries removed - and one extra generator which creates .pb.go with just the structs & not the reflection stuff - would probably do the job. It's possible with just hacks, I think.

gedw99, 2022-09-20:
Anyone found a drop in replacement for xml/ encoding ?

gedw99, 2022-09-20:
> I am a third-party package using the xml package, which is not easy to replace.

Me too

gitslav, 2022-11-10:
I can't believe Go (TinyGo) is about to miss the boat on WASM

gedw99, 2023-03-05:
There is are a few non reflection based xml packages out there to replace encoding/xml.

How did the tinygo Team get JSON done ? I ask because xml is holding me back

dgryski, 2023-03-05:
I'm working on fixing reflection support which should enable both JSON and XML

jptosso, 2023-03-05:
Tinygo+ encoding/xml is now working and the code is pushed

gedw99, 2023-03-05:
@jptosso yhsnk you so much !!

gedw99, 2023-03-05:
I tested XML and works really well ...

i have a dependency on https://github.com/go-pdf/fpdf.
It does  encoding/gob and got the following error
/usr/local/opt/go/libexec/src/encoding/gob/decode.go:597:21: undefined: reflect.MakeMapWithSize

encoding/gob is still in the works right ? Is there anything i can do to help..

dgryski, 2023-03-05:
The latest development branch should have `reflect.MakeMapWithSize`, but I believe `encoding/gob` doesn't work for other reasons.

gedw99, 2023-03-05:
@dgryski thanks for the prompt response.

When I compiled it kick now tell me the line number of where it failed. I assume that I need to use sone flags or not compile removing debugb symbols ?

vhespanha, 2025-01-27:
hey, @dgryski.
do you know why we still can't implement `encoding/gob`? i'd appreciate it a lot if you could link where this was discussed.

dgryski, 2026-10-27:
@vhespanha `encoding/gob` now passes.  The `IgnoreDepthLimit` test needs `reflect.ArrayOf()`, but other than that we seem to be working.

```
~/go/src/github.com/tinygo-org/tinygo $ tinygo version
tinygo version 0.44.0-dev-b7ec945f darwin/arm64 (using go version go1.27.2 and LLVM version 23.1.4)
~/go/src/github.com/tinygo-org/tinygo $ tinygo test -skip='(IgnoreDepthLimit|CountEncodeMallocs|CountDecodeMallocs)' encoding/gob
ok  	encoding/gob	2.144s
```

davecheney, 2026-10-28:
I tried a basic struct round trip with both `encoding/json` and `encoding/gob` using TinyGo 0.44.0-dev-eae3f875. It ran successfully and printed `JSON and gob round trips passed`:

```go
package main

import (
	"bytes"
	"encoding/gob"
	"encoding/json"
	"fmt"
)

type Message struct {
	Name  string
	Count int
}

func main() {
	want := Message{Name: "tinygo", Count: 42}

	data, err := json.Marshal(want)
	if err != nil {
		panic(err)
	}
	var fromJSON Message
	if err := json.Unmarshal(data, &fromJSON); err != nil {
		panic(err)
	}
	if fromJSON != want {
		panic("JSON round trip failed")
	}

	var buf bytes.Buffer
	if err := gob.NewEncoder(&buf).Encode(want); err != nil {
		panic(err)
	}
	var fromGob Message
	if err := gob.NewDecoder(&buf).Decode(&fromGob); err != nil {
		panic(err)
	}
	if fromGob != want {
		panic("gob round trip failed")
	}

	fmt.Println("JSON and gob round trips passed")
}
```

This confirms these basic round trips work; it doesn’t test every serialization feature or third-party library.

closed 2026-10-28 by davecheney