tinygo-org/tinygo #448: TypeError: import object field 'env' is not an Object

talentlessguy, 2019-07-08:
My setup:

**index.html**

```html
<!DOCTYPE html>
<html>
<head>
  <meta charset="utf-8">
  <title>Go WebAssembly App</title>
</head>
<body>
  <script src="wasm_exec.js"></script>
  <script>
    // Polyfill
    if (!WebAssembly.instantiateStreaming) {
      WebAssembly.instantiateStreaming = async (res, obj) => (
        await WebAssembly.instantiate(await (await res).arrayBuffer(), obj)
      )
    }
    const go = new Go()
    WebAssembly.instantiateStreaming(fetch('../build/out.wasm'), go.importObject).then((result) => {
      go.run(result.instance)
      WebAssembly.instantiate(result.module, go.importObject)
    })
  </script>
</body>
</html>
```

**main.go**

```go
package main

import "fmt"

func main() {
	fmt.Println("Hello WASM!")
}
```

Directory structure:

```
.
├── build
│   └── out.wasm
├── glue
│   ├── index.html
│   └── wasm_exec.js
├── src
│   └── main.go
└── Taskfile.yml
```

It compiles fine using builtin `go build`:

```sh
env GOARCH=wasm GOOS=js go build -o build/out.wasm src/main.go
```

But it doesn't work with TinyGo:

```sh
tinygo build -o build/out.wasm -target wasm src/main.go
```

It throws this error:

```js
TypeError: import object field 'env' is not an Object
```

I think something wrong with `go.importObject`, it doesn't receive env. Or js code in html should work in another way?

sendilkumarn, 2019-07-08:
I also faced the same error. Take the `wasm_exec.js` file available in this repo. That fixes it.

talentlessguy, 2019-07-08:
So I need to pick their wasm_exec instead of `GOROOT`'s?

closed 2019-07-08 by talentlessguy