tinygo-org/tinygo #450: Build tag intended for code that can not (yet) be built with TinyGo

peterhellberg, 2019-07-17:
Since there are packages in the standard library that can not be built with TinyGo just yet; I was thinking that there could be a use case for putting code using such packages behind a build tag.

Best case scenario would, probably, be to use existing tags. If that is not possible then to add one used by TinyGo, and then negate it for the code in question.

Unfortunately it seems like TinyGo is currently unable to build code importing [`go/build`](https://golang.org/pkg/go/build/)

TinyGo satisfies the `gc` build tag so it can not (currently) be used. 

> This could possibly be resolved by having the value of the `Compiler` field in `build.Default` changed from `gc` to something else. (That could potentially introduce other issues?)

I'm not really sure about what name to give such a tag, maybe `tinygo`?

peterhellberg, 2019-07-17:
### Default context in `go/build` 
```go
package main

import (
	"fmt"
	"go/build"
)

func main() {
	fmt.Printf("%#v\n", build.Default)
}
```

#### On [The Go Playground](https://play.golang.org/p/MPXWxdBjKAT)
```
build.Context{GOARCH:"amd64p32", GOOS:"nacl", GOROOT:"/usr/local/go", GOPATH:"", CgoEnabled:false, UseAllFiles:false, Compiler:"gc", BuildTags:[]string(nil), ReleaseTags:[]string{"go1.1", "go1.2", "go1.3", "go1.4", "go1.5", "go1.6", "go1.7", "go1.8", "go1.9", "go1.10", "go1.11", "go1.12"}, InstallSuffix:"", JoinPath:(func(...string) string)(nil), SplitPathList:(func(string) []string)(nil), IsAbsPath:(func(string) bool)(nil), IsDir:(func(string) bool)(nil), HasSubdir:(func(string, string) (string, bool))(nil), ReadDir:(func(string) ([]os.FileInfo, error))(nil), OpenFile:(func(string) (io.ReadCloser, error))(nil)}
```

#### On my laptop running Go `1.12.7`
```
build.Context{GOARCH:"amd64", GOOS:"darwin", GOROOT:"/usr/local/Cellar/go/1.12.7/libexec", GOPATH:"/Users/peter/Go", CgoEnabled:true, UseAllFiles:false, Compiler:"gc", BuildTags:[]string(nil), ReleaseTags:[]string{"go1.1", "go1.2", "go1.3", "go1.4", "go1.5", "go1.6", "go1.7", "go1.8", "go1.9", "go1.10", "go1.11", "go1.12"}, InstallSuffix:"", JoinPath:(func(...string) string)(nil), SplitPathList:(func(string) []string)(nil), IsAbsPath:(func(string) bool)(nil), IsDir:(func(string) bool)(nil), HasSubdir:(func(string, string) (string, bool))(nil), ReadDir:(func(string) ([]os.FileInfo, error))(nil), OpenFile:(func(string) (io.ReadCloser, error))(nil)}
```

aykevl, 2019-07-17:
TinyGo has a special build tag: `tinygo`. Also, it supports the `-tags` flag (as of version 0.7.0).

closed 2019-07-17 by aykevl