tinygo-org/tinygo #452: Type alias causes compiler runtime error: invalid memory address or nil pointer dereference

advanderveer, 2019-07-18:
When calling a method on the alias the compile will fail with a runtime error. To reproduce:

in file `project/a/a.go`:
```go
package a

// Attr is an unordered map of attributes
type Attr map[string]interface{}

// Foo returns the map of attributes sorted by key names
func (a Attr) Foo() { println("foo") }
```

in `project/main.go`
```go
package main

import (
	"project/a"
)

type At = a.Attr

func main() {
	a := At{}
	a.Foo()
}
```
Then run inside `project/`:
```
tinygo build -o wasm.wasm -target=wasm .
```

deadprogram, 2019-09-18:
Is this still an issue in the current `dev` branch?

aykevl, 2019-09-18:
I can still reproduce it in the dev branch, so yes this is still an issue.
Smaller/simpler repro:

```go
package main

type Attr int

func (a Attr) Foo() {}

type At = Attr

func main() {
	var a At
	a.Foo()
}
```

jadr2ddude, 2019-09-20:
The reason that this occurs is that the function is not discovered until the `SimpleDCE` pass, where a bug in our handling of discovery prevents it from getting marked properly. It therefore gets deleted immediately. I will hopefully get a fix PR in soon.

jadr2ddude, 2019-09-23:
This should be fixed by #574

jadr2ddude, 2019-10-20:
This appears to have been fixed.

closed 2019-10-20 by jadr2ddude