tinygo-org/tinygo #453: Custom type of empty interface causes compile runtime error: index out of range

advanderveer, 2019-07-19:
When an empty interface type is defined asserting against it crashes the compiler.

```Go
package main

type Foo interface{}

func main() {
	var v Foo

	if _, ok := v.(Foo); ok {
	}
}
```

then run `tinygo build -o wasm.wasm -target=wasm .` (tinygo 0.7.0) to get:

```
panic: runtime error: index out of range

goroutine 1 [running]:
github.com/tinygo-org/tinygo/compiler.(*Compiler).getInterfaceMethodSet(0xc000178000, 0xc0002f0720, 0xc0002d1cc0)
	/Users/distiller/project/compiler/interface.go:213 +0x4be
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseTypeAssert(0xc000178000, 0xc0005a6090, 0xc0003f0000, 0x7b93dc0)
	/Users/distiller/project/compiler/interface.go:256 +0x161
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseExpr(0xc000178000, 0xc0005a6090, 0x6dcdf80, 0xc0003f0000, 0xc0003f0000, 0x1, 0x9)
	/Users/distiller/project/compiler/compiler.go:1868 +0x321
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseInstr(0xc000178000, 0xc0005a6090, 0x6dcdf20, 0xc0003f0000)
	/Users/distiller/project/compiler/compiler.go:1013 +0x15cf
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseFunc(0xc000178000, 0xc0005a6090)
	/Users/distiller/project/compiler/compiler.go:987 +0xe44
github.com/tinygo-org/tinygo/compiler.(*Compiler).Compile(0xc000178000, 0x7ffeefbffa4e, 0x1, 0x1b, 0x0, 0x0)
	/Users/distiller/project/compiler/compiler.go:308 +0xafb
main.Compile(0x7ffeefbffa4e, 0x1, 0x7ffeefbffa37, 0x9, 0xc000142180, 0xc0001a3ee0, 0xc0001a3d48, 0x0, 0x0)
	/Users/distiller/project/main.go:125 +0xa94
main.Build(0x7ffeefbffa4e, 0x1, 0x7ffeefbffa37, 0x9, 0x7ffeefbffa49, 0x4, 0xc0000b1ee0, 0xc000114240, 0x8e00368)
	/Users/distiller/project/main.go:337 +0xe2
main.main()
	/Users/distiller/project/main.go:696 +0xefd
```

deadprogram, 2019-12-24:
Just confirmed that same issue still exists in 0.11.0

aykevl, 2019-12-26:
A fix has been merged into the dev branch, so this will be part of the next release.

closed 2019-12-26 by aykevl