tinygo-org/tinygo #458: Version parsing fails for Go betas

QuLogic, 2019-07-24:
With Go 1.13beta1 installed, tinygo smoketests fail when [parsing the version](https://kojipkgs.fedoraproject.org//work/tasks/3072/36453072/build.log):

> error: could not read version from GOROOT (/usr/lib/golang): failed to parse minor version: strconv.Atoi: parsing "13beta1": invalid syntax

deadprogram, 2019-08-03:
I'm not sure that we support running TinyGo inside anything but the currently released Go, but that seems like something that could be changed. Also useful for testing with the next version of Go.

aykevl, 2019-08-03:
I think it would be useful to support this, for testing newer versions. Patches are welcome :)
And yes, TinyGo only has been tested with Go 1.11 and Go 1.12.

deadprogram, 2019-08-05:
This has been addressed in the `dev` branch, thanks to @QuLogic 

I will close this issue now. Thanks for the PR!

PR #476 (2019-08-05): Fix parsing of beta Go versions.
Note that there are other issues with Go 1.13 that I have yet to investigate. This just fixes the obvious one in the version parsing.

Fixes #458.
closed 2019-08-05 by deadprogram