tinygo-org/tinygo #460: Misleading option `-heap-size` for wasm

lei-april, 2019-07-26:
This options was introduced by commit d627208c48501e649a0c2a477f77556af1e35a99. From what I see, it actually configures size of whole wasm linear memory, not just the "heap".

As regard to wasm bytecodes generated by the LLVM backend, the linear memory consists of three sections: an explicit stack, the data section and the heap. 

From source of wasm-ld:
```cpp
// The default memory layout is as follows, from low to high.
//
//  - initialized data (starting at Config->GlobalBase)
//  - BSS data (not currently implemented in llvm)
//  - explicit stack (Config->ZStackSize)
//  - heap start / unallocated
```

However, the argument of option `-heap-size` seems to apply to the whole linery memory:
```go
if spec.GOARCH == "wasm" {
    // Round heap size to next multiple of 65536 (the WebAssembly page
    // size).
    heapSize := (config.heapSize + (65536 - 1)) &^ (65536 - 1)
    ldflags = append(ldflags, "--initial-memory="+strconv.FormatInt(heapSize, 10))
}
```

So I think it's a little confusing to use the phrase "heap size" here.

deadprogram, 2019-08-03:
Perhaps this should be renamed to something like `memory-size` to more accurately describe what it does. @aykevl what do you think?

aykevl, 2019-08-03:
The original intent was that `-heap-size` would also configure the heap size on *nix systems (using `malloc` to allocate the heap on startup). However, this turned out to be a bit more tricky to implement than I originally expected so I left it out.

Changing to `-memory-size` sounds good.

aykevl, 2021-01-09:
#1567 sidesteps this issue by removing this flag entirely.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram