tinygo-org/tinygo #464: flashing in WSL will not work

perrito666, 2019-07-28:
As a part of the gophercon workshop I tried to use tinyGO with WSL as suggested in the instructions, most of it work but there is a not-so-small blocker.

By trying to flash an `arduino nano 33` provided by @deadprogram at the event I found out that all compilation process working but `bossac` does not complete succesfully:

In Windows `COM#` (serial port number #) ports get mapped to `/dev/ttyS#` for WSL, on my system the nano 33 would be COM3 when connected and COM4 when in flash mode.

The invocation (as made from inside tinygo) `bossac -d -i -e -w -v -R --port=/dev/ttyS4 --offset=0x200 /tmp/tinygo182745706/main.bin`

Fails with:

```
hduran@Perrisurface:~/tinyGoWs/gophercon-2019$ sudo /usr/local/tinygo/bin/bossac -d -i -e -w -v -R --port=/dev/ttyS4 --offset=0x200 main.bin
Send auto-baud
Set binary mode
version()=v2.0 [Arduino:XYZ] Apr 19 2019 14:38:48
Connected at 115200 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
write(addr=0x20004000,size=0x34)

SAM-BA operation failed
```

After experimenting with several changes on the pseudo linux system and the underlying windows I did not manage to make `bossac` like it, there are mentions in several forums for windows devs that some instructions will not work yet (or ever) so not all proper serial operations work.

After a suggestion by @justinclift in the slack channel I tried to flash something in arduino IDE and captured the invocation made there, which resulted in:

`C:\Users\perri\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe -i -d --port=COM4 -U true -i -e -w -v C:\Users\perri\AppData\Local\Temp\arduino_build_394539/Blink.ino.bin -R`

The first part of the folder would change depending on your arduino IDE installation, in my case I used the downloadable installer from the site so it resulted in `%localappdata%/Arduino` (where `%localappdata%` is a system variable) but arduino ide can also be installed via the Windows App store, and the resulting path, I think, is ` %userprofile%\documents`

With this information I tried to invoke arduino's bundled bossac with the resulting command:

` /mnt/c/Users/perri/AppData/Local/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac.exe -d -i -e -w -v --port=COM4 main.bin`  where main.bin is the output of invoking ` tinygo build -target arduino-nano33 -port=$NANO33_DEV_PATH -o main.bin ./sensor/arduino/step0/main.go` and the main.go file is the one from the gophercon samples to try with a known working sample.
Please note that given that I invoked the windows bossac I could use `COM4` as the port

Now, what could be done here is asking the user to set an env variable with the path to `bossac.exe` and have a way to tell to tinygo that it's running on windows, the only blocker I found is that being an older version of bossac it does not take `--offset` so it is not a trivial replace.

Feel free to contact me to discuss this and test it as much as necessary on windows (or to dismiss it entirely not sure how many users actually use WSL I just decided to check it since it was one of the provided options)

Sadly I am not familiar enough with tinygo as to be able to provide a patch.

deadprogram, 2019-08-03:
Hi @perrito666 it might be worthwhile to try the Arduino fork of BOSSA:

https://github.com/arduino/BOSSA

I see some fixes for building in Windows that might work for you? Please give a try if you can.

For a number of important reasons, such as newer board definitions, I really do not want to have to support older versions of BOSSA.

perrito666, 2019-08-04:
Ill try this afternoon if life doesn’t get in the middle

perrito666, 2019-08-05:
ok, so partial attempt, I build this version in wsl2 and tried to run the classic instructions, did not work (failed with the same `SAM-BA` error) after work ill try to figure out how to build this in windows (i seem to lack some of the tooling to build an actual windows binary using gnu tools so ill have to deal with that before trying), but I foresee that this should work so the only missing piece would be a way to specify a path to bossac including binary (which will be named bossac.exe) in order to invoke the windows programmer and instructions to build bossac in windows which ill happily provide once I figure them out.

trevrosen, 2019-08-21:
@perrito666 really looking forward to this if you can do it -- would be huge for adoption!

perrito666, 2019-08-21:
I did not forget about it i was interrupted by a two week work trip in which i did not take my windows laptop sorry :( upon return ill try to figure how yo compile the suggested bossac for windows and relay the info

deadprogram, 2019-08-22:
I just realized that on this page there are MSI installers for Windows, both 64-bit and 32-bit:
https://github.com/shumatech/BOSSA/releases

Seems to me that you should try this @perrito666 before continuing the effort to build yourself from source. If this works, and you can run the `tinygo build` command, then seems to me that we've got a solution...

perrito666, 2019-08-22:
@deadprogram ill give it a shot (somewhere in the weekend after reuniting with my windows computer) i am wondering if this will work despite not being the https://github.com/arduino/BOSSA/releases fork for BOSSA

deadprogram, 2019-08-22:
I have no idea, really. Hard to tell from the commit log if any of the commits from the Arduino fork are needed here. Have to try it, I think.

perrito666, 2019-08-24:
Ok the good news is that the windows 1.9.1 version of bossac does something, the bad news is that what it does seems to be erasing the arduino and never finishes programming it so it bricks it, ill see if i can revert this, so far no luck from within arduino ide, the device turns on but is not recognized bu the computer at all. I suppose the arduino patch is of some relevance, ill see to fetch a couple more of these for more tests and in the mean time figure a way to compile the windows version of the arduino tuned fork and try to make it available, sorry for the new roadblock

deadprogram, 2019-08-25:
Ouch, sorry it has bricked the board. Hopefully you can get it back to life.

I will also try to do some investigation on this in the next week.

perrito666, 2019-08-25:
No worries it happens, i wonder if we can talk to whoever maintains the parallel version in arduino to see if they can cut us a compiled version

deadprogram, 2019-08-27:
I wonder if the issue is https://github.com/arduino/ArduinoCore-samd/issues/417

deadprogram, 2019-08-27:
via https://github.com/shumatech/BOSSA/issues/99

Oliviajhon890, 2019-09-03:
I've­­ made15,000usd­­ so ­­far­­ this ­­year­­ working ­­0nline. I'm ­­using ­­an online­­ business ­­opportunity. My­­ lovely ­­friend­­ showed me and­­ I've­­ made ­­lot of­­ advantage­­ of­­ this ­­site. It's­­ really­­ user ­­friendly­­ and ­­I'm just so happy­­ that­­ I­­ found­­ out­­ about­­ it.. .. Thanks a lot

Check it out for yourself >HERE →→  w­­­­w­­­­w­­­­.­­­­s­­­­e­­­­c­­­­u­­­­r­­­­e­­­­w­­­­o­­­­r­­­­k­­­­9­­­­9­­­­.­­­­c­­­­o­­­­mⒶ

Don't include this ☞ Ⓐ☜ in web link THANKS

deadprogram, 2019-10-24:
I was able to use the bossac Windows pre-built binary as described here: https://tinygo.org/getting-started/windows/#arduino-nano33-iot

I think that we can close this issue now, let me know what you think @perrito666 thank you.

perrito666, 2019-10-24:
That is fantastic :D yeah this is enough for closing, it makes me super happy, looking forward to traveling to a place where i can purchase one of these to keep trying

closed 2019-10-24 by perrito666