tinygo-org/tinygo #477: How can I upload an object file to the Arduino using LLVM clang?

tdh8316, 2019-08-05:
Hi, As I know, It needs to convert the object file to a hex file. But I cannot convert the object file generated by LLVM to a hex file using `llvm-objcopy`.

Whenever I try this:
```cmd
> llvm-objcopy -O ihex Blink.cpp.o Blink.cpp.hex -B avr
llvm-objcopy.exe: error: Invalid output format: 'ihex'.
```

The object file is generated by this command:
```cmd
>> clang -S -emit-llvm -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -x c++ -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I "C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" Blink.cc -o ./Blink.cc.ll -I "C:\Program Files (x86)\Arduino\hardware\tools\avr\avr\include" --target=avr
```

Hence I wonder if there's another way to convert an object file to Intel Hex for Arduino, or if there's a different procedure that the Tinygo uses.

aykevl, 2019-08-05:
I think you mean a fully linked executable when you say "object file"?

In TinyGo, there is some code to do this in the compiler itself without using external tools. This avoids relying on external dependencies that may or may not be installed.
The `llvm-objcopy` command is still a work in progress, I would suggest using `avr-objcopy` instead.

https://github.com/tinygo-org/tinygo/blob/7d5542dda73e8dda066b604870eac279cd644a7e/objcopy.go#L94-L96

Closing this as it is not an issue in TinyGo. Feel free to ask more.

closed 2019-08-05 by aykevl

tdh8316, 2019-08-05:
I am sorry for this issue not related to Tinygo. :-/

I was searching for how can I upload the LLVM IR to the Arduino.
And I found this project and found out this exactly do what I want to.

`avr-objcopy` did not work for me
`avr-objcopy -O ihex Blink.cpp.o B
link.cpp.hex -B avr
avr-objcopy:Blink.cpp.o: File format not recognized`

aykevl, 2019-08-05:
> `clang -S -emit-llvm`

This emits textual LLVM IR, of course this is not going to work. What you need:

1. Use Clang like any other compiler, leaving out the `-S -emit-llvm` flags. Or if you really must use LLVM IR, the `llc` tool may be of use.
2. Use `avrdude` to upload a hex file to the Arduino.

You can enable verbose logging in the Arduino IDE to see the exact commands you should use for uploading.
Also, if there is no good reason to use Clang, I would recommend to use the GNU toolchain which is way more stable and reliable (and most likely produces better code as well). The reason TinyGo uses LLVM is that it really is built on top of it and can't reasonably switch to a different backend without a full rewrite.