tinygo-org/tinygo #504: Cannot compile a program on Arduino Uno

talentlessguy, 2019-08-11:
I try to compile this program and flash it into my Arduino Uno.

Arduino Uno is located in `/dev/ttyUSB0`.

Here's the error message:

```sh
Desktop/Coding/arduino via 🐹 v1.12.7 
➜ tinygo flash -target arduino -port /dev/ttyUSB0 main.go
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: cannot find -lm
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo234022434/main: exit status 1
```

Go program:

```go
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 1000)

		led.High()
		time.Sleep(time.Millisecond * 1000)
	}
}
```

`go env`:

```
GOARCH="amd64"
GOBIN=""
GOCACHE="/home/v1rtl/.cache/go-build"
GOEXE=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOOS="linux"
GOPATH="/home/v1rtl/go"
GOPROXY=""
GORACE=""
GOROOT="/usr/lib/go-1.12"
GOTMPDIR=""
GOTOOLDIR="/usr/lib/go-1.12/pkg/tool/linux_amd64"
GCCGO="gccgo"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD=""
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build354841673=/tmp/go-build -gno-record-gcc-switches"
```

`go version`:

```
go version go1.12.7 linux/amd64
```

`tinygo version`:

```
tinygo version 0.7.1 linux/amd64
```

aykevl, 2019-08-11:
Have you installed `avr-libc`?

talentlessguy, 2019-08-12:
> Have you installed `avr-libc`?

No, but I have `gcc-avr` installed :thinking: 

Okay, I installed `avr-libc`, now I get this error:

```
/bin/sh: 1: avrdude: not found
error: failed to flash /tmp/tinygo800734584/main.hex: exit status 127
```

aykevl, 2019-08-12:
> No, but I have `gcc-avr` installed ￼

I know, the naming of these packages isn't great. Also, `gcc-avr` doesn't depend on the libc it uses (`avr-libc`) which you would normally expect. You have to install both manually. I have made a PR to the documentation to more prominently include this information, see https://github.com/tinygo-org/tinygo-site/pull/38 (I copied your error messages, I hope that's okay).

To fix the other error, simply install `avrdude`.

talentlessguy, 2019-08-12:
It works now but docs doesn't cover it. So I close the issue in favor of https://github.com/tinygo-org/tinygo-site/pull/38

closed 2019-08-12 by talentlessguy