tinygo-org/tinygo #506: panic: runtime error: index out of range

tegk, 2019-08-16:
Using Ubuntu 19.04 amd64 and installed the .deb

I can not compile my hello world :-(

tinygo build -o main -target linux-amd64 ./test.go

```
panic: runtime error: index out of range

goroutine 1 [running]:
main.LoadTarget(0x7ffd5277f414, 0xb, 0xc0000a5d60, 0x95e40d, 0x34a3cc0)
	/home/circleci/project/target.go:204 +0x95c
main.Build(0x7ffd5277f420, 0x9, 0x7ffd5277f407, 0x4, 0x7ffd5277f414, 0xb, 0xc0000a5ee0, 0xc00005e690, 0xffffffffffffffff)
	/home/circleci/project/main.go:332 +0x3f
main.main()
	/home/circleci/project/main.go:696 +0xefd

```

aykevl, 2019-08-16:
TinyGo should not panic here. But you can simply leave out the target and it should compile for the host:

    tinygo build -o main ./test.go

Alternatively, you can pass a full triple, such as `linux--amd64` (note the extra dash).

deadprogram, 2019-08-25:
I dot not think that `-target linux-amd64` is a valid target. In any case, we should probably validate that flag and return a more graceful error.

deadprogram, 2019-09-18:
Just committed PR #570 that validates the `target` flag.

closed 2019-09-18 by tegk