tinygo-org/tinygo #508: Closures and channel operations result in a bug

aykevl, 2019-08-16:
The following doesn't produce the expected output:

```go
package main

var strch = make(chan string)

func main() {
	strch := strch // comment this line to avoid the bug
	go func() {
		println("sending...")
		strch <- "hello"
		println("sent!")
	}()
	println(<-strch)
}
```

There appears to be some sort of interaction between closures and channel operations that goes wrong.

jadr2ddude, 2019-08-30:
It appears that we are accidentally passing in `undef` for the context. As a result, the closure ends up with a channel pointer that actually points to something else (when tested against #513, I ended up with `ch=0x401802 <runtime.fakeCoroutine>`).

jadr2ddude, 2019-08-30:
And it does this because it is treated as a static call, and therefore we never actually do anything with the closure context.

jadr2ddude, 2019-10-18:
As it turns out, this fails to compile with assertions enabled.

aykevl, 2019-11-02:
Fixed by #640.

PR #640 (2019-11-02): Fix miscompile of static calls to closures
This PR fixes #508

Previously, spawning a goroutine with a closure whose function is statically known would result in a call that did not pass in the closure context. This resulted in undefined behavior. After copying in the appropriate context extraction code, the function-lowering pass attempted to transform the `makeGoroutine` operation, which caused the code to compile but have an LLVM assertion failure. The relevant code in the function-lowering pass has been modified to ignore `makeGoroutine` calls.
closed 2019-11-02 by aykevl