tinygo-org/tinygo #510: Buffered Channels

jadr2ddude, 2019-08-17:
After looking in the SSA, I realized that we currently ignore the buffering parameter of a channel. Channel buffering is commonly used in order to implement high-level concurrency constructs such as semaphores. This would probably require some modification to garbage collection to recognize whatever buffer we use.

aykevl, 2019-08-17:
You have a good point there. Ignoring it now probably gets more code to compile (somewhat incorrectly) but we should definitely add support for buffered channels.

With the conservative GC, we shouldn't need any modifications as it considers everything to be a potential pointer.

aykevl, 2019-08-17:
This is a good resource when it comes to implementing channels: https://www.youtube.com/watch?v=KBZlN0izeiY
TinyGo might do things a bit differently, but buffered channels should probably be implemented with the same ring buffer mechanism.

deadprogram, 2019-09-22:
PR #513 that implements this has now been merged into the `dev` branch, so I am going to close this issue. Please re-open if needed, thanks everyone!

closed 2019-09-22 by deadprogram