tinygo-org/tinygo #512: Teensy: 4.0 support

wdevore, 2019-08-17:
Hi, is there any effort in progress or planned for supporting the **Teensy 4.0** board?
It uses the *ARM Cortex-M7*.

Thanks.

aykevl, 2019-08-17:
Hi @wdevore, there is no support right now but multiple people have requested it.
Step one is this PR: https://github.com/posborne/cmsis-svd/pull/82

deadprogram, 2019-09-17:
That PR was merged into the cmsis-svd repo, so now it can be brought into TinyGo to generate the device wrapper.

firelizzard18, 2020-01-22:
In #863, I tweaked the generator, so `src/device/nxp/mimxrt1062.go` is now generated, but there's a lot of work to do, most of which I am unfamiliar with

jbpratt, 2020-11-23:
I think this can be closed, there is now a `teensy40` target!

deadprogram, 2020-11-24:
Indeed! Now closing.

closed 2020-11-24 by deadprogram