tinygo-org/tinygo #518: proposal: rewrite super-short sleep operations to non-blocking operations

jadr2ddude, 2019-08-26:
I am not entirely sure if this is a good idea or not.

Consider the following line of code:
```
time.Sleep(time.Microsecond)
```

Right now, this code would yield to the scheduler. This could take much longer than intended.

For short time periods, we could either:
1. emit `nop` instructions with inline assembly
2. call a precision timing function which could be written in assembly

One of the issues with this is that there really is no perfect threshold when going to the scheduler makes sense.

Alternatively, we could make another function for this.

aykevl, 2019-08-31:
I think this would be an improvement. One microsecond may still be too long (I think it's possible to switch to a different goroutine and back in that time) but if the time to switch is equal or longer than the time to sleep it is definitely better to just sleep directly without scheduling.

At the same time, programmers may have the expectation that `time.Sleep` will always cause a context switch which may be relevant in CPU-intensive operations that would otherwise block other goroutines from running. I don't think `time.Sleep` provides any such guarantees so I don't think this should be a real concern - `runtime.Gosched` should be used instead.

jadr2ddude, 2019-09-04:
@aykevl After thinking this over, I do not think that this is the appropriate solution. It might actually make more sense to add a separate timing function (`machine.Delay` maybe?) to do this, so that we do not end up merging these two behaviors.