tinygo-org/tinygo #525: Support for the WiiPy/LoPy boards from PyCom

prologic, 2019-08-29:
Would it be possible (_and is anyone interested?_) in adding support for the WiPy/LoPy boards from [Pycom](https://pycom.io)?

prologic, 2019-08-29:
See the [WiPy 3.0 specs](https://pycom.io/product/wipy-3-0/)

jadr2ddude, 2019-08-29:
So, it appears that this board uses the ESP32. The ESP32 uses an ISA called xtensa, which is not yet supported in LLVM. The main thing stopping us at this point is the xtensa patch set being merged into LLVM:

https://esp32.com/viewtopic.php?t=11412#p46446

Once that is in, it should be feasible to add ESP32 support, and therefore be able to run on this board.

prologic, 2019-08-30:
Cool that sounds exciting; Look forward to it. Shall we keep this issue open until we see traction on the LLVM front?

aykevl, 2019-08-31:
I'm looking forward to Xtensa support in LLVM. Take a look here to see the latest discussion: https://reviews.llvm.org/D64830 - it might take a while to integrate it in LLVM.

Until then, I'm afraid there's not a whole lot we can do right now so closing. As soon as Xtensa support is merged, this issue should be reopened.

closed 2019-08-31 by aykevl