tinygo-org/tinygo #527: WASM: fatal error: stack overflow

88250, 2019-08-31:
node.go: 

```go
package main

type Node interface {
	Parent() Node
}

type BaseNode struct {
	parent          Node
}

func (n *BaseNode) Parent() Node {
	return n.parent
}

func (n *BaseNode) SetParent(parent Node) {
	n.parent = parent
}
```

main.go in package main:

```go
package main

import (
	"fmt"
)

//go:export md
func md(markdownText string) string {
	fmt.Println("TinyGo！")
	var node Node // use interface Node not work
	//var node *BaseNode // use struct *BaseNode works
	node = &BaseNode{}
	_ = node
	return "TinyGo！"
}

func main() {
	select {}
}
```

build script: 

```
tinygo build -o lute.tinygo.wasm -target wasm
```

error log:

```
runtime: goroutine stack exceeds 1000000000-byte limit
fatal error: stack overflow

runtime stack:
runtime.throw(0x34a8749, 0xe)
        /usr/local/go/src/runtime/panic.go:617 +0x72
runtime.newstack()
        /usr/local/go/src/runtime/stack.go:1041 +0x6f0
runtime.morestack()
        /usr/local/go/src/runtime/asm_amd64.s:429 +0x8f

goroutine 1 [running]:
runtime.newobject(0x345bc00, 0x0)
        /usr/local/go/src/runtime/malloc.go:1067 +0x51 fp=0xc024000338 sp=0xc024000330 pc=0x8acf81
go/types.TypeString(0x35251e0, 0xc0015db7d0, 0x0, 0x0, 0xc0004ce000)
[... 183 lines cut ...]
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x35251e0, 0xc0015db7d0, 0x1, 0xc0029586c0)
        /home/circleci/project/compiler/interface.go:117 +0x760 fp=0xc024009d98 sp=0xc024009be8 pc=0xbab4f0
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x3525260, 0xc0015db860, 0x1, 0xc0029586b0)
        /home/circleci/project/compiler/interface.go:133 +0x294 fp=0xc024009f48 sp=0xc024009d98 pc=0xbab024
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x35251e0, 0xc0015db7d0, 0x1, 0xc0029586a0)
        /home/circleci/project/compiler/interface.go:117 +0x760 fp=0xc02400a0f8 sp=0xc024009f48 pc=0xbab4f0
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x3525260, 0xc0015db860, 0x1, 0xc002958690)
        /home/circleci/project/compiler/interface.go:133 +0x294 fp=0xc02400a2a8 sp=0xc02400a0f8 pc=0xbab024
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x35251e0, 0xc0015db7d0, 0x1, 0xc002958680)
        /home/circleci/project/compiler/interface.go:117 +0x760 fp=0xc02400a458 sp=0xc02400a2a8 pc=0xbab4f0
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x3525260, 0xc0015db860, 0x1, 0xc002958670)
        /home/circleci/project/compiler/interface.go:133 +0x294 fp=0xc02400a608 sp=0xc02400a458 pc=0xbab024
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x35251e0, 0xc0015db7d0, 0x1, 0xc002958660)
        /home/circleci/project/compiler/interface.go:117 +0x760 fp=0xc02400a7b8 sp=0xc02400a608 pc=0xbab4f0
github.com/tinygo-org/tinygo/compiler.getTypeCodeName(0x3525260, 0xc0015db860, 0x1, 0xc002958650)
        /home/circleci/project/compiler/interface.go:133 +0x294 fp=0xc02400a968 sp=0xc02400a7b8 pc=0xbab024

goroutine 5 [syscall]:
os/signal.signal_recv(0x0)
        /usr/local/go/src/runtime/sigqueue.go:139 +0x9c
os/signal.loop()
        /usr/local/go/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
        /usr/local/go/src/os/signal/signal_unix.go:29 +0x41
```

aykevl, 2019-08-31:
This bug has been fixed in the dev branch: https://github.com/tinygo-org/tinygo/pull/474
Duplicate issue: #309.

We will probably have a release in the near future which includes this fix.

closed 2019-08-31 by aykevl

88250, 2019-08-31:
> This bug has been fixed in the dev branch: #474
> Duplicate issue: #309.
> 
> We will probably have a release in the near future which includes this fix.

Thanks, it is very exciting!