tinygo-org/tinygo #528: WASM: panic: runtime error: invalid memory address or nil pointer dereference

88250, 2019-08-31:
I am using dev branch (git commit 8a5fa51f6023ef78a7e33767a06d68ddea89dfdb).
test code is here https://github.com/b3log/lute/tree/b45b26b13ac291e955af2aeb6d6ab22330e06a04/wasm

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb53f4e]

goroutine 1 [running]:
github.com/tinygo-org/tinygo/vendor/golang.org/x/tools/go/ssa.(*Type).Type(...)
        /root/go/src/github.com/tinygo-org/tinygo/vendor/golang.org/x/tools/go/ssa/ssa.go:1465
github.com/tinygo-org/tinygo/compiler.(*Compiler).getRuntimeType(...)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:438
github.com/tinygo-org/tinygo/compiler.(*Compiler).getLLVMRuntimeType(0xc00016a000, 0x338a1fe, 0xb, 0xbdc4e0)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:445 +0xae
github.com/tinygo-org/tinygo/compiler.(*Compiler).makeStructTypeFields(0xc00016a000, 0xc00126d1d0, 0xc00133ec30)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:96 +0x55
github.com/tinygo-org/tinygo/compiler.(*Compiler).getTypeCode(0xc00016a000, 0x34111c0, 0xc00126d1d0, 0x2b)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:72 +0x54d
github.com/tinygo-org/tinygo/compiler.(*Compiler).getTypeCode(0xc00016a000, 0x34110c0, 0xc00126d1a0, 0x33)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:60 +0x4da
github.com/tinygo-org/tinygo/compiler.(*Compiler).getTypeCode(0xc00016a000, 0x3411100, 0xc004312720, 0x1)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:64 +0x390
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseMakeInterface(0xc00016a000, 0x7f0464649a98, 0x3411100, 0xc004312720, 0x0, 0x0)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:27 +0xb3
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseExpr(0xc00016a000, 0xc005ada360, 0x3418a80, 0xc004597950, 0xc004597950, 0x1, 0x0)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:1656 +0x242c
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseInstr(0xc00016a000, 0xc005ada360, 0x3418a20, 0xc004597950)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:1043 +0x169c
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseFunc(0xc00016a000, 0xc005ada360)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:1017 +0xe44
github.com/tinygo-org/tinygo/compiler.(*Compiler).Compile(0xc00016a000, 0x3386b95, 0x1, 0x1b, 0x0, 0x0)
        /root/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:336 +0xa3d
main.Compile(0x3386b95, 0x1, 0x7fffe01141e2, 0x10, 0xc0001284e0, 0xc00606fed0, 0xc00606fd30, 0x0, 0x0)
        /root/go/src/github.com/tinygo-org/tinygo/main.go:131 +0xb1b
main.Build(0x3386b95, 0x1, 0x7fffe01141e2, 0x10, 0x7fffe01141fb, 0x4, 0xc000095ed0, 0xc0000f8220, 0xffffff)
        /root/go/src/github.com/tinygo-org/tinygo/main.go:343 +0xe2
main.main()
        /root/go/src/github.com/tinygo-org/tinygo/main.go:704 +0xf6c
```

jadr2ddude, 2019-08-31:
I tried running that code on my machine, and I got some very weird but different errors. I ran into an infinite recursion issue due to a type cycle in `regexp/syntax`:

```
[]*regexp/syntax.Regexp
**regexp/syntax.Regexp
*regexp/syntax.Regexp
regexp/syntax.Regexp
struct{Op regexp/syntax.Op; Flags regexp/syntax.Flags; Sub []*regexp/syntax.Regexp; Sub0 [1]*regexp/syntax.Regexp; Rune []rune; Rune0 [2]rune; Min int; Max int; Cap int; Name string}
```

With debugging turned off, I instead got an error from `interp`:
```
error: interp: branch on a non-const-propagated constant expression
```

My current go version is 1.12.7, and I am on TinyGo commit f64ee0596609cc93222d3851186e5691bb9b74dc. I suspect your issue may be specific to how we process a certain version of a stdlib package. Can you please state what `go version` you are on?

88250, 2019-09-01:
Thanks for the try to use my code.  `go version go1.12.7 linux/amd64`. My code use builds conditions: `-tags wasm`, I use tinygo-build.sh in the same directory.

88250, 2019-09-01:
@jadr2ddude I have removed some dependencies to stdlib. Please test my code on master branch if you have time.

closed 2019-09-04 by 88250