tinygo-org/tinygo #531: reflect Comparable undefined

tobowers, 2019-09-03:
This might be related to: https://github.com/tinygo-org/tinygo/issues/530 ?

I'm getting an error directly from the Go src:

```
# tinygo build -o wasm.wasm -target=wasm .
# context
../../../../../usr/local/go/src/context/context.go:471:26: reflect.TypeOf(key).Comparable undefined (type reflect.Type has no field or method Comparable)
```

This is using the docker image: tinygo/tinygo:0.7.1 sha256:59dd8b425cdf69e9fbca3876060073fb0f37d8fa6fb53cad2ef2cc3c18202c07

I'll try to get a super simple reproducible.

tobowers, 2019-09-03:
Yeah here we go:

```
package main

import (
	"context"
	"fmt"
)

func main() {
	ctx := context.Background()
	fmt.Println(ctx)
}
```

deadprogram, 2019-09-03:
Hi @tobowers as you have already surmised, this code is already in the `dev` branch as added in https://github.com/tinygo-org/tinygo/pull/490

Now closing, thanks!

closed 2019-09-03 by deadprogram