tinygo-org/tinygo #538: Tinygo for windows and mac

pjebs, 2019-09-08:
Can I use tiny go to build small applications intended to be run on Mac and windows? (i.e. I'm after the small binary size for my use case)

jadr2ddude, 2019-09-08:
We do in fact support Mac OS. However, windows is significantly different and, at least with what we have now, supporting Windows is definitely non-trivial.

pjebs, 2019-09-09:
Can it execute other applications from `exec` package?

pjebs, 2019-09-09:
I don't see any instructions on how to build _for_ Mac. I can see how to install tinygo on mac for developing on certain boards.

deadprogram, 2019-09-10:
Probably just compiling on macOS without any `-target` will choose that OS. However, I think very little is currently implemented for the runtime on macOS.

aykevl, 2019-09-10:
Yes, you can leave out the `-target` on macOS to build for that platform. In fact, that's done in CI to make sure this continues to work.
At the moment, it's not possible to build an application for macOS on a different OS, but I hope this will become possible in the future by somehow binding to their libc. At least the definitions [appear to be open source](https://opensource.apple.com/source/Libsystem/Libsystem-1252.250.1/) so we could theoretically work with that.

At the moment, there is no support for Windows at all. This is an entirely different beast (not *nix such as Linux and macOS) so will take a bit more work.

> Can it execute other applications from `exec` package?

I don't think the `"os/exec"` package will work in TinyGo at the moment.

QuLogic, 2019-09-26:
> you can leave out the `-target` on macOS to build for that platform. In fact, that's done in CI to make sure this continues to work.

Is that still true though? As I noted [here](https://github.com/tinygo-org/tinygo/pull/426#discussion_r303227346), there is no smoke test that doesn't pass a `-target`. The only one that doesn't pass it explicitly, uses a `.wasm` extension which [passes a target implicitly](https://github.com/tinygo-org/tinygo/blob/master/main.go#L699).

pjebs, 2019-09-26:
For my use case I only need the `os/exec` package to work for mac and ideally windows. I don't think tinygo is appropriate for my project. Feel free to close issue. Thank you for your time.

closed 2019-09-26 by deadprogram

aykevl, 2019-09-26:
> Is that still true though?

Yes. It's not tested in the smoke tests, but in the normal `go test` fashion:

https://github.com/tinygo-org/tinygo/blob/cc313798f381532e4874a7166fa3c216234c005d/main_test.go#L48-L53