tinygo-org/tinygo #549: OTA?

11azars, 2019-09-14:
Hey! I'm somewhat new to TinyGo, but very excited by its potential. I am interested in exploring some sort of OTA setup with TinyGo, and was wondering if there is any existing work in this area. I have little understanding of how TinyGo actually works on these devices (or more accurately LLVM does), but I assume this would look like having a program overwrite its own flash and force a reset? 

Obviously it's far more complex than this, but I would love to get a discussion going around this.... I may be able to start building something out if I get a few good pointers in the right direction (I'm a web/big-data/systems guy, microcontrollers are more of a hobby sadly).

aykevl, 2019-09-14:
Hi @11azars, glad you're interested!
At the moment, there is no support for OTA in any form. I've been thinking a bit about it but haven't actually done anything. One of the blockers is actually having a way to communicate over the air, but there is the [espat driver](https://github.com/tinygo-org/drivers/tree/master/espat) which can do that now.

There are two main ways OTA can be implemented:
1. Scratch space for the new application is reserved in the internal flash. OTA is done by first copying the new image over to the scratch space (while running the old application), then resetting to a small bootloader that overwrites the old application with the new application in the scratch space, and resets to start the new application. This requires more flash but can have a better UI.
2. To start OTA, reboot into a bootloader which is able to set up a connection to download the new firmware image, directly overwriting the old application. This requires far less flash space, but is harder to get right and must always be completed before it is possible to start the new application.

You always need some sort of bootloader, as it is not possible to overwrite the currently executing flash and it is _usually_ not possible to start the application from a different address (though it might be possible).

I once wrote a bootloader for MicroPython that never got merged (and I decided that I wanted to change the protocol), you can take a look here if you're interested: https://github.com/aykevl/micropython/tree/nrf-dfu-mbr-wip (this one follows approach 2 with in-place updates).

11azars, 2019-09-14:
Thanks for the info! I'm definitely interested in exploring this further.
My usecase would involve downloading the new image over LoRa packet radio.
That being said, in an ideal world, OTA would be accessible using an
io.Writer. does a bootloader package exist?

My work would probably be to expose a bootloader API to the application,
and let the application handle image retrieval. Do you think this is in the
realm of possibility?

deadprogram, 2024-11-18:
Closing due to age. Thanks to everyone!

closed 2024-11-18 by deadprogram