tinygo-org/tinygo #566: Error on macOS

dangreen, 2019-09-17:
```sh
inygo build -o main.wasm -target wasm ./main.go
# unicode/utf8
../../../../../../usr/local/Cellar/go/1.13/libexec/src/unicode/utf8/utf8.go:28:8: expected ';', found b00000000 (and 10 more errors)
```

aykevl, 2019-09-17:
Unfortunately, Go 1.13 is not yet supported. Please try Go 1.12 until we've added support for Go 1.13.

dangreen, 2019-09-17:
@aykevl Thanks! 

Can you help with another question? There are any solution to use WaitGroup ?

```
../github.com/disintegration/imaging/utils.go:28:14: WaitGroup not declared by package sync
```

closed 2019-09-17 by dangreen

jadr2ddude, 2019-09-17:
@dangreen currently it is possible to implement similar functionality using channels

If you have any more questions, I would suggest joining the `tinygo` channel on [Gophers Slack](https://invite.slack.golangbridge.org/).