tinygo-org/tinygo #571: Policy: Regression Testing

jadr2ddude, 2019-09-18:
I would like to propose that we add regression tests every time we submit a bug fix. This could mean one of the following:
1. tweak an existing `testdata/` sample to hit the edge case
2. add a file in a subsection (`testdata/regression` maybe?) for testing the case

I suspect that, especially in the long term, this could cut the amount of time spent debugging.
An especially good candidate for trying this out could be #526.

deadprogram, 2019-09-18:
I do not #526 is a good candidate as the code in question only executes on actual SAMD21 hardware itself.

jadr2ddude, 2019-09-18:
Well, not entirely. I actually mostly meant #532, which is a subset of #526.

aykevl, 2019-09-18:
Ideally, I fully agree.
In practice, this is kind of difficult with all the various hardware involved. I certainly try to add tests whenever possible, but it's not always easy to do. Even in the case of #532 (sleep queue corruption bug) I wouldn't really know how to add a test. For bugs like #526, I see that there is https://jumper.io/ but I'm guessing that is going to cost a lot of money (unless they're open to supporting open-source software) - if it even emulates this chip close enough. Also, I'm more strict on it now than I was in the past, with the result that a lot of existing code isn't well tested. If you think a PR needs more tests, please point that out to keep us all alert on it.

closed 2019-09-18 by jadr2ddude